nkolban/esp32-snippets #287: has no member named 'srvc_id'

maxpromer, 2017-12-17:
```
C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                                              ^

In file included from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:9:0:

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                                                                                                        ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:131:118: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

    BLERemoteCharacteristic* pBLERemoteCharacteristic = pBLERemoteService->getCharacteristic(BLEUUID(evtParam->notify.char_id.uuid));

                                                                                                                      ^

In file included from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:9:0:

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:133:111: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'char_id'

     ESP_LOGE(LOG_TAG, "Could not find characteristic with UUID %s for notification", BLEUUID(evtParam->notify.char_id.uuid).toString().c_str());

                                                                                                               ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168: note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

                                                                                                                                                                        ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:196:55: error: no matching function for call to 'BLEUUID::BLEUUID(esp_gatt_id_t&)'

    BLEUUID uuid = BLEUUID(evtParam->search_res.srvc_id);

                                                       ^

In file included from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteCharacteristic.h:18:0,

                 from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteService.h:16,

                 from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.h:18,

                 from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:14:

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:26:2: note: candidate: BLEUUID::BLEUUID()

  BLEUUID();

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:26:2: note:   candidate expects 0 arguments, 1 provided

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:25:2: note: candidate: BLEUUID::BLEUUID(esp_gatt_srvc_id_t)

  BLEUUID(esp_gatt_srvc_id_t srcvId);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:25:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:24:2: note: candidate: BLEUUID::BLEUUID(uint8_t*, size_t, bool)

  BLEUUID(uint8_t* pData, size_t size, bool msbFirst);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:24:2: note:   candidate expects 3 arguments, 1 provided

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:23:2: note: candidate: BLEUUID::BLEUUID(esp_bt_uuid_t)

  BLEUUID(esp_bt_uuid_t uuid);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:23:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_bt_uuid_t'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:22:2: note: candidate: BLEUUID::BLEUUID(uint32_t)

  BLEUUID(uint32_t uuid);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:22:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint32_t {aka unsigned int}'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:21:2: note: candidate: BLEUUID::BLEUUID(uint16_t)

  BLEUUID(uint16_t uuid);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:21:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'uint16_t {aka short unsigned int}'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:20:2: note: candidate: BLEUUID::BLEUUID(std::__cxx11::string)

  BLEUUID(std::string uuid);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:20:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(const BLEUUID&)

 class BLEUUID {

       ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'const BLEUUID&'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:18:7: note: candidate: constexpr BLEUUID::BLEUUID(BLEUUID&&)

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEUUID.h:18:7: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'BLEUUID&&'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:197:94: error: no matching function for call to 'BLERemoteService::BLERemoteService(esp_gatt_id_t&, BLEClient*)'

    BLERemoteService* pRemoteService = new BLERemoteService(evtParam->search_res.srvc_id, this);

                                                                                              ^

In file included from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.h:18:0,

                 from C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src\BLEClient.cpp:14:

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteService.h:29:2: note: candidate: BLERemoteService::BLERemoteService(esp_gatt_srvc_id_t, BLEClient*)

  BLERemoteService(esp_gatt_srvc_id_t srvcId, BLEClient* pClient);

  ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteService.h:29:2: note:   no known conversion for argument 1 from 'esp_gatt_id_t' to 'esp_gatt_srvc_id_t'

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteService.h:27:7: note: candidate: BLERemoteService::BLERemoteService(const BLERemoteService&)

 class BLERemoteService {

       ^

C:\Users\Max\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLERemoteService.h:27:7: note:   candidate expects 1 argument, 2 provided
```

i use `BLE_server`. I try fix it but i found "struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'" is true.

https://github.com/espressif/arduino-esp32/blob/81a9c45a1ef5833e80c9235ff393a7641c669a94/tools/sdk/include/bluedroid/esp_gattc_api.h#L163

[image]

nkolban, 2017-12-18:
Howdy @maxpromer 
I agree that the notify data structure does not have a property called `srvc_id`.  Now we get to the core of the question.  Your issue report showed a fragment of source code that includes:

```
 BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));
```

so my question becomes .... where does that source code come from?  Where did you get that code that is being compiled?

maxpromer, 2017-12-18:
Thank for reply. @nkolban 

This source code, me download from release 0.4.2 in https://github.com/nkolban/ESP32_BLE_Arduino/releases

i recheck `BLEClient.cpp` via WinRAR and i found it.

[image]

nkolban, 2017-12-18:
Oh Wow!!!  Thank you kind sir.  It turns out that we have been updating the ESP32_BLE_Arduino repository but NOT flagging new releases as ... well ... releases.  All we have been doing is uploading new code into Master when what we should have been doing is uploading new code into master AND flagging new releases.  I just tagged a new release at 0.4.7.

Looking in the latest code, I can't seem to find any further references.

maxpromer, 2017-12-18:
I update to release v0.4.7 and all examples it work 👍 

Thank you very much for do it libraries and fast reply.

closed 2017-12-18 by maxpromer