nkolban/esp32-snippets #288: Crucial question about BLEClient class

chegewara, 2017-12-17:
Im wondering, why we have handled ESP_GATTC_OPEN_EVT event instead ESP_GATTC_CONNECT_EVT:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L181-L191

nkolban, 2017-12-17:
In the BLEClient code we have a method called "connect".  Its purpose is to connect to a remote BLE Server.  We achieve this by calling the underlying ESP-IDF function called `esp_ble_gattc_open()`.  Since we are performing in an asynchronous environment, we block the call until we get a corresponding `ESP_GATTC_OPEN_EVT` which is the partner to `esp_ble_gattc_open()`.  Your question asks ... as I understand it ... "Why are you watching for `ESP_GATTC_OPEN_EVT` and not `ESP_GATTC_CONNECT_EVT`?".  The glib and simple answer is that when we open a connection using esp_ble_gattc_open() we get an `ESP_GATTC_OPEN_EVT` as a response and not an `ESP_GATTC_CONNECT_EVT`.

However, this does open up a very salient question for which I have no answer.  Since we see the existence of an event call `ESP_GATTC_OPEN_EVT` and another event called `ESP_GATTC_CONNECT_EVT` ... what then are the semantic distinctions between each one?  Under what circumstances do I receive `ESP_GATTC_OPEN_EVT` and under what circumstances do I get an `ESP_GATTC_CONNECT_EVT` and what do each mean?   I have no idea and suggestions/study are welcome

chegewara, 2017-12-17:
Research in progress...

chegewara, 2017-12-20:
`Close a virtual connection to a GATT server. gattc maybe have multiple virtual GATT server connections when multiple app_id registed, this API only close one virtual GATT server connection. if there exist other virtual GATT server connections, it does not disconnect the physical connection. if you want to disconnect the physical connection directly, you can use esp_ble_gap_disconnect(esp_bd_addr_t remote_device).`

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gattc.html#_CPPv219esp_ble_gattc_close13esp_gatt_if_t8uint16_t

closed 2018-01-05 by chegewara