nkolban/esp32-snippets #290: BLECharacteristic ESP_GATTS_READ_EVT issue

chegewara, 2017-12-17:
Ive been trying to write ble hid keyboard application and i discovered issue in function handling ESP_GATTS_READ_EVT event. Everything works fine unless im trying to connect hid keyboard to laptop with windows 10. It was long battle but finaly i discovered the issue. In normal case usage we having default mtu set to 23, but what windows is doing is negotiating MTU and and we are having MTU 517. Now what function is doing is sending only 22 bytes, since windows can receive 516 or 517 bytes (dont know for sure) windows is thinking that he received whole message and is sending request for next message. 
What we need to do is keep tracking MTU value when we receive ESP_GATTS_MTU_EVT event and slice message according to this value.

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L370

nkolban, 2017-12-17:
I think you are probably right and a ton more study needs to be performed to figure out the correct solution.  Specifically, imagine we have a characteristic of length *L* bytes.  When a request to read the characteristic arrives, how do we respond to it?  Can we send *L* bytes?  If not, do we send *R* bytes where *R* is less than *L* in which case what should the value of *R* be?  What does a follow on request look like? ... the list goes on and on.  We need to study the theory of BLE and then map that to the practical of ESP32.

chegewara, 2017-12-17:
This is logging from my app. What we can see here is:
- we have ESP_GATTS_CONNECT_EVT event, we can set mtu here because parameter passed with this event does not have info about MTU,
- next event we have is ESP_GATTS_MTU_EVT and is set to value 517

```
D (159785) BLEUtils: GATT ServerEvent: ESP_GATTS_CONNECT_EVT
D (159787) BLEUtils: [conn_id: 0, remote_bda: 80:a5:89:82:f5:d2]
D (159793) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
V (159800) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdc20c), owner: <N/A>
V (159807) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdc3c8), owner: <N/A>
V (159815) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdcefc), owner: <N/A>
V (159822) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdd0e8), owner: <N/A>
V (159829) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdd3b8), owner: <N/A>
V (159837) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdd5a4), owner: <N/A>
V (159844) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdded0), owner: <N/A>
V (159852) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffde0c8), owner: <N/A>
V (159859) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffde66c), owner: <N/A>
V (159866) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdec04), owner: <N/A>
V (159874) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdee44), owner: <N/A>
D (159881) BLEServer: << handleGATTServerEvent
D (159886) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_MTU_EVT
D (159893) BLEUtils: GATT ServerEvent: ESP_GATTS_MTU_EVT
D (159899) BLEUtils: [conn_id: 0, mtu: 517]
D (159903) BLEServer: >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
D (159909) BLEServer: << handleGATTServerEvent

```

Of course we can negotiate MTU somewhere when we are connected, or we can assume default value described in bluetooth specification and change it when we receive MTU negotiation event.

chegewara, 2017-12-17:
My quick fix helped me to install hid keyboard on windows 10. Keyboard is rcognised and installed. Now i have to find out where i have another issue.

chegewara, 2017-12-26:
Im reopenning this because we need to do more research and fix some procedures according to BLUETOOTH SPECIFICATION Version 4.2 [Vol 3, Part F] 3.4.2 and following.

closed 2018-01-01 by chegewara