nkolban/esp32-snippets #291: Handle white-list processing

nkolban, 2017-12-18:
While reading this book ... http://shop.oreilly.com/product/0636920033011.do

I was reading about "White Lists" ... and was then looking in the ESP32 docs for related APIs and found

* http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv228esp_ble_gap_update_whitelistb13esp_bd_addr_t
* http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv230esp_ble_gap_get_whitelist_sizeP8uint16_t

Since we have APIs at the ESP-IDF that are exposed ... let us now investigate how we might expose and use these through the C++ classes.

chegewara, 2017-12-18:
Another step in ble security

nkolban, 2017-12-18:
New code/functions added:

* `BLEAdvertising::setScanFilter()`
* `BLEDevice::whiteListAdd()`
* `BLEDevice::whiteListRemove()`

No breaking changes and no changes in existing apps introduced by these functions.

closed 2017-12-18 by nkolban