nkolban/esp32-snippets #292: BLEClient crashes on Disconnect (lost connection) with server

chegewara, 2017-12-18:
Case:
- flash esp32 with SampleServer with esp32 with SampleClient
- perform connection
- unplug esp32 with SampleServer app from power
- esp32 SampleClient crashes with:

```
E (16913) BLERemoteCharacteristic: Disconnected
abort() was called at PC 0x400f4887 on core 0
0x400f4887: std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > >::do_put(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, long double) const at ??:?

Backtrace: 0x4008917f:0x3ffdd720 0x400891ab:0x3ffdd740 0x400f4887:0x3ffdd760 0x400f48ce:0x3ffdd780 0x400e63d7:0x3ffdd7a0 0x40101ec3:0x3ffdd7c0 0x400d3ad5:0x3ffdd820 0x40105e14:0x3ffdd940
0x4008917f: invoke_abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:572

0x400891ab: abort at /home/esp32/esp/esp-idf/components/esp32/./panic.c:572

0x400f4887: std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > >::do_put(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, long double) const at ??:?

0x400f48ce: std::istreambuf_iterator<char, std::char_traits<char> >::equal(std::istreambuf_iterator<char, std::char_traits<char> > const&) const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_classes.tcc:114 (discriminator 6)

0x400e63d7: std::bad_alloc::~bad_alloc() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/bad_alloc.cc:28

0x40101ec3: BLEService::setHandle(unsigned short) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544

0x400d3ad5: _fiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:33

0x40105e14: is_wifi_clk_peripheral at /home/esp32/esp/esp-idf/components/driver/./periph_ctrl.c:200

```

chegewara, 2017-12-18:
I cant understand why this line is in SampleClient:
`0x40101ec3: BLEService::setHandle(unsigned short) at /home/esp32/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544`

By investigating code we can see that only place it is invoked is here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEService.cpp#L293-L299

But this means we are trying to create service in client application, does that make sens?

nkolban, 2018-01-04:
I'm going to suggest that this is likely a problem with "where" one runs monitor.  In your story, I sense TWO ESP32s ... each running a different app.   Lets say we have ESP32 A running App A and ESP32 B running App B.  It is important that we run "make monitor" in the directory for App A pointing to ESP32 A and a second "make monitor" in the directory App B pointing to ESP32 B.  If we mix them up ... for example ... running "make monitor" in the directory for App A pointing to ESP32 B or running "make monitor" in the directory for App B pointing to ESP32 A then any stack traces will likely decode against the wrong application and produce strange results (perhaps as shown here).  Can you remember/verify where "make monitor" was being run against which instance of the ESP32?

chegewara, 2018-01-04:
Always make monitor is run from app directory with command: `make flash monitor`

closed 2018-01-04 by chegewara