nkolban/esp32-snippets #296: Unable to singlestep BLE code (xtensa-esp32-elf-gdb crashes)

BobaTrek, 2017-12-19:
I am amazed at how rapidly this BLE library is growing and well it is supported - Thanks!

I am considering switching my embedded control project which requires BLE over to the ESP32.  I am using the xtensa tool chain on a Windows 10 computer, Eclipse, xtensa-esp32-elf-gdb, and OpenOCD.  My target is the ESP32 WROVER kit for now since I wanted to make sure I could do single stepping using the built-in JTAG.

Everything was proceeding well until I started including and calling the BLE functions.  Ever since then (about 2 weeks now) I have been unable to single step once I hit a breakpoint.  Sometimes I can do one singlestep, maybe two, but I always end up hitting a breakpoint in another thread (usually queue.c) that I did not set.

This other individual is experiencing the same thing and posted an issue back in October:
[https://github.com/espressif/openocd-esp32/issues/21](url)

I have tried various combinations of hitting a breakpoint, removing all breakpoints, using Step into, over, and out of; run to line, inserting a new breakpoint at the next line and hitting continue, etc.

The only thing that works is to hit a breakpoint, remove all breakpoints, and hit continue.  Then while the program is running again, insert a new breakpoint.

In my research I found some people working on the Cortex-M3 had a similar problem which they solved using this technique:
[http://openocd.org/doc/html/GDB-and-OpenOCD.html#Configuring-GDB-for-OpenOCD](url) which was to add the following to their gdb setup:
    define hook-step
    mon cortex_m maskisr on
    end
    define hookpost-step
    mon cortex_m maskisr off
    end

I tried the same technique for the ESP32 but it did not work.

I have two questions:

1)  Are others seeing this problem as well?

2)  Does anyone know how to work around this (besides printf style debugging)?

This issue is really hurting my efforts to switch over to the ESP32.

nkolban, 2017-12-19:
Up until recently I had studiously avoided JTAG and had managed to get away with simple inserting of debug statements in the log.  I presume I got away with this because by and large I had only been writing snippets of application code.  However, that changed recently when I started to work on a large existent code base of another project that is thousands of lines of code.  I too needed a source level debugger running against the ESP32.  I followed the recipes to get JTAG running against the ESP32 and also am coming up against "wake ups" in other threads that I haven't started to study yet.  They sound identical (or at least similar) to the story you present.

However ... there is nothing in the JTAG/ESP32 story that should be affected by the libraries found in this repository.  To start a thread of discussion/issue here trying to solve a JTAG puzzle is not going to be a great way to proceed.  I am merely a (poor) JTAG consumer and know nothing about GDB, openocd or JTAG technologies.

What I'd suggest is creating a forum post on the ESP32 forum found here:

https://esp32.com/index.php

ideally, search through it first to see if someone has walked this path previously.  If you explain the puzzle as clearly as possible, others with more skills and techniques might be able to assist us.  In addition, it is likely where others will go in the future should/when they have the same problem.  Trying to resolve ESP32 and JTAG as an issue in this repository (C++ utilities for ESP32) is likely a poor direction.

chegewara, 2017-12-19:
From screenshot posted in this issue https://github.com/espressif/openocd-esp32/issues/21 we can see crash is in queue.c which is a part of freeRTOS. If you have some time and you are willing to play some with settings i would suggest to:
- turn off all option related to dual core im menuconfig,
- increase stack size for all tasks
- turn on highest level heap tracing in menuconfig ->heap memory debbuging->heap memory corruption->comprehensive

BobaTrek, 2017-12-19:
Thanks all.  I did originally post this question in the esp32 forums a couple times, but had yet to get any responses.

Today however I did get a response that has solved the issue.

A couple others on the BLE C++ threads had mentioned to me that they also saw the problem, so I will reference the solution here as well.

The problem for me was that the documentation for the ESP WROVER KIT here is incorrect:
[http://esp-idf.readthedocs.io/en/latest/get-started/get-started-wrover-kit.html](url)
On this page, in "Setup Options" for the JTAG jumper JP8 it says to install 5 jumpers to enable JTAG.  However, apparently the 5th jumper labelled S_TDI should not be installed.

JP8 should only have 4 jumpers installed.  @sharif351 informed me of this and included a picture of the correct jumper configuration over here:
[https://github.com/espressif/openocd-esp32/issues/21](url)

I have been happily singlestepping ever since!

(I do still occasionally hit breakpoints (that I did not set) in queue.c, but the debugger no longer crashes at that point and I can usually step my way back into the thread I was trying to step through.)