nkolban/esp32-snippets #297: Error logged when client sends long (>20 bytes) value to characteristic

rweller59, 2017-12-19:
environment:  Eclipse + ESPIDF + C++.
Have ESP set up as ble server. When long characteristic value is written by connected client (I tested up to about 130 char string), callback works fine and entire message is read on characteristic.getValue() call. This is great because I thought I was going to have to manage receiving in 20 byte junks.

However, I get the following error logged - it doesn't seem to have any impact that I have discovered yet.
BT: GATTS_SendRsp conn_id: 3  waiting for op_code = 00
BT: Sending response failed

chegewara, 2017-12-19:
When you last time updated esp32-snippets? Can you pastebin some code example to tests?

rweller59, 2017-12-19:
pretty much the same code we were discussing last week (#261). just getting back to work on the incoming ble portion. everything was updated last week (espidf, toolchain, cpp_utils).

// process BLE connections
class bleServerCallback: public BLEServerCallbacks {
	void onConnect(BLEServer* pServer) {
		ESP_LOGI(LOG, "Bluetooth client connected");
		ble_connected = true;
		}
	void onDisconnect(BLEServer* pServer) {
		ESP_LOGI(LOG, "Bluetooth client disconnected");
		ble_connected = false;
		pServer->getAdvertising()->start();
		}
};

// process external characteristic calls
class bleCharacteristicCallback: public BLECharacteristicCallbacks {
	void onWrite(BLECharacteristic* pCharacteristic) {
		std::string msg = pCharacteristic->getValue();
		ESP_LOGI(LOG, "BLE received:  %s", msg.c_str());
		}

	void onRead(BLECharacteristic* pCharacteristic) {
		ESP_LOGE(LOG, "client read characteristic");
		}
};

// create BLE server, services and characteristics
void init(void) {
	BLEDevice::init("ESP32");
	BLEServer *pServer = BLEDevice::createServer();
	pServer->setCallbacks(new bleServerCallback());
	BLEService *pService = pServer->createService(SERVICE_UUID);

	// characteristic for configuration with callback defined
	configCharacteristic = pService->createCharacteristic(
		BLEUUID(CONFIG_CHARACTERISTIC_UUID),
		BLECharacteristic::PROPERTY_READ   |
		BLECharacteristic::PROPERTY_WRITE  |
		BLECharacteristic::PROPERTY_NOTIFY |
		BLECharacteristic::PROPERTY_INDICATE
	);
	configCharacteristic->addDescriptor(new BLE2902());
	configCharacteristic->setCallbacks(new bleCharacteristicCallback());

	dataCharacteristic = pService->createCharacteristic(
		BLEUUID(DATA_CHARACTERISTIC_UUID),
		BLECharacteristic::PROPERTY_READ   |
		BLECharacteristic::PROPERTY_WRITE  |
		BLECharacteristic::PROPERTY_NOTIFY |
		BLECharacteristic::PROPERTY_INDICATE
	);
	dataCharacteristic->addDescriptor(new BLE2902());

	pService->start();
	pServer->getAdvertising()->start();
}

chegewara, 2017-12-19:
We have new functions implemented in BLE (MTU in this case) which im not 100% sure about being. But quick test with SampleSecureServer + nRF connect on android = works good to me. Im writing 27 bytes long array: 0x112233445566778899212233445566778899312233445566778899

```
D (371142) BLEUtils: GATT ServerEvent: ESP_GATTS_WRITE_EVT
D (371145) BLEUtils: [conn_id: 0, trans_id: 9, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, offset: 0, need_rsp: 1, is_prep: 1, len: 18]
D (371157) BLEUtils: [Data: 112233445566778899212233445566778899]
D (371163) BLEServer: >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
D (371169) BLEValue: >> addPart: length=18
D (371173) BLECharacteristic:  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
D (371184) BLECharacteristic:  - Data: length: 18, data: 112233445566778899212233445566778899
D (371193) BLEServer: << handleGATTServerEvent
D (371198) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
D (371206) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (371211) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (371217) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (371223) BLEServer: << handleGATTServerEvent
D (371228) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
D (371236) BLEUtils: GATT ServerEvent: ESP_GATTS_WRITE_EVT
D (371241) BLEUtils: [conn_id: 0, trans_id: 10, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, offset: 18, need_rsp: 1, is_prep: 1, len: 9]
D (371253) BLEUtils: [Data: 312233445566778899]
D (371258) BLEServer: >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
D (371264) BLEValue: >> addPart: length=9
D (371268) BLECharacteristic:  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
D (371279) BLECharacteristic:  - Data: length: 9, data: 312233445566778899
D (371286) BLEServer: << handleGATTServerEvent
D (371292) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
D (371299) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (371305) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (371310) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (371317) BLEServer: << handleGATTServerEvent
D (371321) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_EXEC_WRITE_EVT
D (371330) BLEUtils: GATT ServerEvent: ESP_GATTS_EXEC_WRITE_EVT

```
Then im reading from server and all seems to be fine.

```
D (378252) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_READ_EVT
D (378252) BLEUtils: GATT ServerEvent: ESP_GATTS_READ_EVT
D (378254) BLEUtils: [conn_id: 0, trans_id: 12, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, is_long: 0, need_rsp:1]
D (378264) BLEServer: >> handleGATTServerEvent: ESP_GATTS_READ_EVT
D (378270) BLECharacteristic: - Testing: 0x2a == 0x2a
I (378276) BLECharacteristic: 23
I (378279) BLECharacteristic: 22
D (378283) BLECharacteristic: Sending a response (esp_ble_gatts_send_response)
D (378291) BLECharacteristic:  - Data: length=22, data=11223344556677889921223344556677889931223344, offset=0
D (378301) BLEServer: << handleGATTServerEvent
D (378306) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
D (378313) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (378320) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (378324) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (378331) BLEServer: << handleGATTServerEvent
D (378336) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_READ_EVT
D (378343) BLEUtils: GATT ServerEvent: ESP_GATTS_READ_EVT
D (378349) BLEUtils: [conn_id: 0, trans_id: 13, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, is_long: 1, need_rsp:1]
D (378359) BLEServer: >> handleGATTServerEvent: ESP_GATTS_READ_EVT
D (378365) BLECharacteristic: - Testing: 0x2a == 0x2a
I (378370) BLECharacteristic: 23
I (378374) BLECharacteristic: 22
D (378378) BLECharacteristic: Sending a response (esp_ble_gatts_send_response)
D (378386) BLECharacteristic:  - Data: length=5, data=5566778899, offset=22
D (378392) BLEServer: << handleGATTServerEvent
D (378398) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
D (378405) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (378411) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (378416) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (378423) BLEServer: << handleGATTServerEvent

```

rweller59, 2017-12-20:
I'll update and try again

rweller59, 2017-12-20:
interesting. I updated cpp_utils and the onWrite method doesn't get called with a long or short value. Re-installed previous version and back to where I was: onWrite gets triggered with long and short values.

chegewara, 2017-12-20:
Now i see what you are telling to me. We are having some logging issue here, because characteristic seems to be written with proper value but  CharacteristicCallbacks::onWrite() is giving us junk:

```
D (45476) BLEUtils: [conn_id: 0, trans_id: 2, bda: 78:1f:db:c0:c7:f3, handle: 0x2a, offset: 0, need_rsp: 1, is_prep: 0, len: 16]
D (45488) BLEUtils: [Data: 737369643a4f72616e67652d38463534]
D (45493) BLEServer: >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
D (45500) BLECharacteristic: >> setValue: length=16, data=737369643a4f72616e67652d38463534, characteristic UUID=0d563a58-196a-48ce-ace2-dfec78acc814
D (45513) BLECharacteristic: << setValue
D (45517) BLECharacteristic:  - Response to write event: New value: handle: 2a, uuid: 0d563a58-196a-48ce-ace2-dfec78acc814
D (45528) BLECharacteristic:  - Data: length: 16, data: 737369643a4f72616e67652d38463534
I (45536) SampleServer: 73 73 69 64 
```
Last line is from onWrite() event callback. We should have get 16 characters but we see only 4. Investigating.

rweller59, 2017-12-20:
I was actually using android app and sending String. I will switch over to nRF for further testing. However, with the previous version of cpp_utils, value (string) was successfully read whether it was short (<20 bytes) or long (> 20 bytes). In either case, the onWrite event was triggered once and the entire value was read. When the value was > 20 bytes, I would get the error ( GATTS_SendRsp conn_id: 3 waiting for op_code = 00)

chegewara, 2017-12-20:
is this error generated by esp32-snippets library or by esp-idf?
`GATTS_SendRsp conn_id: 3 waiting for op_code = 00`

Can you just paste whole line with timestamp?

rweller59, 2017-12-20:
I am still on the previous version of cpp_utils since the latest did not trigger the onWrite event. I cannot find the source of the message but not sure what files to look in.

E (131192) BT: GATTS_SendRsp conn_id: 3  waiting for op_code = 00

E (131196) BT: Sending response failed

chegewara, 2017-12-20:
Its a esp-idf error. It would be better if you open issue in esp-idf or on forum. I can do nothing with that.

rweller59, 2017-12-20:
Updated cpp_utils and it seems to work with most recent espidf.
Switched to testing with nRF (sending text). Works exactly like the android app. Messages short and long are received but I get the error code with the long message. Funny that you are not - could our log settings be different?

I will post on esp forum. Do you know whether or not the ESPIDF is intended to be able to receive a long value? Have searched without luck.

chegewara, 2017-12-20:
Esp-idf dont have problems with sending/receiving long messages. It needs to be implemented in controller and it is implemented in cpp_utils. Only thing you can change in menuconfig is `compiler options->optimization level-> release`. We have observed some kind bugged behaviour with debug setting.

rweller59, 2017-12-20:
thanks. posted to eps forum.

rweller59, 2017-12-21:
A bit more research and I realized that I need to request an increased MTU from the client. Doing this using nRF Connect and the ESP onWrite error went away when receiving a value > 20 bytes so this appears to be the source of the original issue. Unfortunately, after requesting larger MTU with android app, the ESP onWrite no longer fires with either short or long values.

I realize that this is an issue in the app code but, it seems pretty simple. I do get an acknowledgement back from the esp when the mtu increase is requested but the request status is false. Thought maybe you had some experience on the android side of things and could offer suggestions. I haven't discovered anything on the esp side of things where I could figure out why the request was failing.

chegewara, 2017-12-21:
Without code or at least comprehensive loggings i can not offer much. As you can see with my test everything works fine. That does not mean im saying there is no bug or something, just i cant recreate this issue. Maybe you can post  sample data you are trying to send. Also check if onWrite proper parameters.

closed 2017-12-31 by rweller59