nkolban/esp32-snippets #298: Suggested enhancement for cpp_utils/json

rweller59, 2017-12-20:
I keep adding these changes every time I update cpp_utils so hoping we can just add to library...

json.cpp
/**
 * @brief Convert the JSON object to an unformatted string.
 * @return A JSON string representation of the object.
 */
std::string JsonArray::toStringUnformatted() {
	char *data = cJSON_PrintUnformatted(m_node);
	std::string ret(data);
	free(data);
	return ret;
} // toStringUnformatted

/**
 * @brief Convert the JSON object to an unformatted string.
 * @return A JSON string representation of the object.
 */
std::string JsonObject::toStringUnformatted() {
	char *data = cJSON_PrintUnformatted(m_node);
	std::string ret(data);
	free(data);
	return ret;
} // toStringUnformatted

json.j (class JsonObject and JsonArray)
	std::string toStringUnformatted();

nkolban, 2017-12-20:
To hear is to obey.  Changes added.  Please review and if delighted close out the issue ... and if not, let us know what we can change next.

rweller59, 2017-12-20:
Works for me. thanks for quick response.

closed 2017-12-20 by rweller59