nkolban/esp32-snippets #299: Connection between NRF52832 IC and ESP32

saitejamanchu, 2017-12-20:
Hello, I want to establish a connection between NRF52832 IC and ESP32 Module.
I implemented BLE on ESP 32 and achieved communication between Nordic Mobile app and ESP32.
However I am unable to establish the connection between ESP32 and NRF52832 IC.

My setup is as follows 1) ESP32 as peripheral with Nordic UART Service 
2) NRF52832 as central running ble_app_uart_c, this is example code with Nordic UART service

Please help me in this regard.

nkolban, 2017-12-21:
Many thanks for taking the time to post about the resolution.

closed 2017-12-21 by nkolban

ursamir, 2019-10-14:
@saitejamanchu Can you share me your code base for this. I am working on an interesting concept.
The idea is programming Argon(Particle) ESP32 from Nordic nRF52840 which is connected to PC using debugger(particle) with a CMSIS-DAP and DAPLink interface.

The reason is, there is no other programming interface for ESP32 onboard.

-S@M