nkolban/esp32-snippets #300: i2c_master_stop: rc=263 Timeout

rweller59, 2017-12-20:
This is probably and ESPIDF issue but, again, since I'm using the cpp_utils library, I am mentioning here.

Environment:  Eclipse, C++, ESPIDF.

Connected to 4 I2C sensors. Communications in general works without error. However, occasionally (once an hour and then at increasing frequency) I get this error and it seems to appear more frequently the longer the app runs.

nkolban, 2017-12-21:
Are there any patterns to it?  For example, does the error occur against one type of I2C slave device?  What kinds of I2C slaves are these?  What is the last I2C API that was executed that timed-out?  Are the "wire" lengths in spec.  If the wires get too long or noisy then bits get lost and the protocol will fail.

rweller59, 2017-12-21:
It is vary haphazard. Lengths are in spec but built on proto board. I am putting on solder board in the next couple of days to eliminate any potential connection issues. The reason I posted this now is that things run fine for long periods of time (hours) than I start getting the error which seems to increase in frequency once its started.  I will try to narrow it down a bit after soldering and re-post then.

rweller59, 2017-12-30:
After getting rid of jumpers and soldering everything, this issue seems to have disappeared.

closed 2017-12-30 by rweller59