nkolban/esp32-snippets #301: BLEAdvertising add functionality request

chegewara, 2017-12-20:
As for now we can change only advertising data by setting custom data but we have no option to change advertising parameters. Sometimes its required to change advertising timings or advertising type or even we would like to turn on private mode by setting address type random. Now we are forced to use "default" values.

nkolban, 2017-12-21:
Lets review this item in context with the previous #196.  Let's see what *meta* properties of a server performing advertising we want to expose.  Maybe if we can make a list here we can approach the implementation task in one sitting.  As we study BLE and we find parameters that affect the nature of advertising we can list them here.

chegewara, 2017-12-23:
Lately we have found that some peripheral device can advertise data and we can read data without connecting to peripheral. What if we would like esp32 become such kind peripheral? Im thinking there is too much permutations to implement it (or maybe not) but we have flag fields and by adding rawData field we could cover all options, leaving the user field to construct raw data deppend to needs.

closed 2018-01-01 by chegewara