nkolban/esp32-snippets #304: opcode: 0x2016

d3cline, 2017-12-21:
I receive a disconnection sometimes. I have tried modifying tx power but I do not believe that is it. 
I am using Android studio and some custom kotlin code to use onCharacteristicRead() When it fails the Android Studio error is, 

{{{
W/BluetoothGatt: onCharacteristicRead() - Device=24:0A:C4:10:A6:E6 handle=42 Status=133
}}}

And this error shows in the serial monitor on Arduino IDE, despite me having serial disabled which leads me to believe it is something in the ESP's IDE or elsewhere above Arduino.

{{{
E (167860) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x2016
E (227907) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x2016
E (349419) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x2016
}}}

I am not sure where to start. I realize it may be something in my arduino code or design, but I am not sure where to start adding verbosity or modifying the sdkconfig file.

chegewara, 2017-12-21:
With this error code `BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x2016` its best to open issue here:
https://github.com/espressif/arduino-esp32/issues
or here https://github.com/espressif/esp-idf/issues (but here devs may be angry ;) )
This message is related to lower level stack and we can do nothing or very few

d3cline, 2017-12-21:
as silly as it is and very ESPcentric I added a delay someplace and fixed the bug I was having this time. It somewhat helped to increase verbosity to debug as that thread you linked me to asks because I was able to 'see' the stack trace and 'get a feel' for what I thought might be the issue which was speed or the cpu needing a break. Not a very scientific fix I suppose, but there you have it.

closed 2017-12-21 by d3cline