nkolban/esp32-snippets #306: BLE_server doesn't compile !!

dev-ahmed-ibrahim, 2017-12-22:
When I try to compile, the following errors appear ,

In file included from C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLEDescriptor.h:16:0,

                 from C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLECharacteristic.h:16,

                 from C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLEServer.h:19,

                 from C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLEDevice.h:18,

                 from C:\Program Files (x86)\Arduino\libraries\Bluetooth\examples\BLE_notify\BLE_notify.ino:21:

C:\Program Files (x86)\Arduino\libraries\Bluetooth/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type

   pthread_mutex_t   m_pthread_mutex;

   ^

In file included from C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLEDevice.h:20:0,

                 from C:\Program Files (x86)\Arduino\libraries\Bluetooth\examples\BLE_notify\BLE_notify.ino:21:

C:\Program Files (x86)\Arduino\libraries\Bluetooth/BLEUtils.h:50:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

                                                 ^

What can I do to solve this problem ?

Thanks ,,,,

nkolban, 2017-12-22:
Howdy my friend, lets see if we can't get a solution going here.

First, in Arduino land there are MANY permutations at work ... it is a stack built on a stack built on a stack and things have to line up just right.

I am assuming you are working with the Arduino IDE ... if yes:

1. What platform (Windows, Linux etc)
2. What release of the Arduino IDE (the latest as of this date is 1.8.5)

Once one has the Arduino IDE installed, one must then install the Arduino-ESP32 drivers that are found here:

https://github.com/espressif/arduino-esp32

Installation instructions can be found here ...

https://github.com/espressif/arduino-esp32#installation-instructions

Since I haven't seen any "version" identifier for Arduino-ESP32, we won't be able to tell if you are back-level or not ... since it doesn't do any harm to simply re-download and re-install ... lets go ahead and do this.

At this point, you will have Arduino IDE and the Arduino-ESP32 libs.  Now let us pause here.

I see from your report that you have the BLE C++ libraries installed at:

C:\Program Files (x86)\Arduino\libraries\Bluetooth

I want you to delete that directory (saving the contents if you need to keep).

You will now find that the BLE C++ libraries are **already** part of the latest Arduino-ESP32 environment.  These were packed as part of the latest release of the Arduino-ESP32 distribution.

Lets see how far we get with this story.  If there are still problems, post back and we'll pick up the thread again.

CGB01, 2018-03-06:
Hi Neil, I was *SUPER* excited to see the was now a BLE library for the Arduino IDE environment...  Just had to try it.  Sadly, I ran into the same problem @dev-ahmed-ibrahim 

My environment is;
Windows 7 Professional 64 bits
Arduino 1.8.5
Adafruit HUZZAH32 ESP32 Board

I have pulled the latest Arduino-ESP32 distribution and added to the Arduino library version 0.4.7 of ESP BLE Arduino.

When I compile code from Andreas Spiess (BLE_Proximity_Sensor.ino) I get the following errors;
---------------
In file included from C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDescriptor.h:16:0,

                 from C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLECharacteristic.h:16,

                 from C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEServer.h:19,

                 from C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:18,

                 from C:\Users\Claude\Documents\Arduino\TestCode\BLE_Proximity_Sensor\BLE_Proximity_Sensor.ino:20:

C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/FreeRTOS.h:46:3: error: 'pthread_mutex_t' does not name a type

   pthread_mutex_t   m_pthread_mutex;

   ^

In file included from C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:20:0,

                 from C:\Users\Claude\Documents\Arduino\TestCode\BLE_Proximity_Sensor\BLE_Proximity_Sensor.ino:20:

C:\Users\Claude\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEUtils.h:50:49: error: 'esp_gattc_service_elem_t' has not been declared

  static std::string gattcServiceElementToString(esp_gattc_service_elem_t *pGATTCServiceElement);

                                                 ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Claude\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board Adafruit ESP32 Feather.
---------------

Any help would be appreciated.

Thanks,
-Claude.

chegewara, 2018-03-06:
Hi @CGB01,
try to update esp-idf in Arduino folder by running this file
C:\Users\Claude\Documents\Arduino\hardware\espressif\esp32\tools\get.exe

CGB01, 2018-03-07:
@chegewara Yes, as part of updating my arduino-esp I also did the "get.exe" command.

chegewara, 2018-03-07:
@CGB01 can you check this file in your system:
https://github.com/espressif/arduino-esp32/blob/master/package/package_esp32_index.template.json

CGB01, 2018-03-07:
Yes it is there but not the same as on GitHub.  So now the question I have is why did it not get updated when I pulled an update from Git GUI?

chegewara, 2018-03-07:
@CGB01 Yes, thats why  asked you to check it. You have old toolchain and most likely old esp-idf installed with you arduino-ide

CGB01, 2018-03-08:
@chegewara I will backup the old directory structure and pull a new one.  I'll let you know how that works for me.

PS: Thanks for your assistance :)

EDIT:  Just finished pulling a new Arduino-esp32 from GitHub and running the "get" command afterwards and PRESTO!  All seem to be good.  Compiled the BLE_Proximity_Sensor sketch without a glitch and it runs fine.

closed 2018-11-13 by chegewara