nkolban/esp32-snippets #308: BLE c++ classes fail to compile if BLE server is disabled

robpelu, 2017-12-22:
Hello I'm trying what you proposed about mixing C++ and C code in the ESP32 and works as you said.

The issue I have while using your BLEcpp clases is that I don't want to enable BLE server as my ESP will only be client, mostly because the size of the binary is too large to fit in the partition (I'm using OTA so 1MB partition).

So if I disable BLE server from the menuconfig, the compilation fails.

BR, Robert.

CC build/wpa_supplicant/src/fast_crypto/fast_sha256.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa-debug-module/trax.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/esp32-ota-https.elf
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDevice.o):(.literal._ZN9BLEDevice15gapEventHandlerE22esp_gap_ble_cb_event_tP22esp_ble_gap_cb_param_t+0x8): undefined reference to `esp_ble_gap_security_rsp'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDevice.o):(.literal._ZN9BLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x38): undefined reference to `esp_ble_gatts_register_callback'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDevice.o):(.literal._ZN9BLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x3c): undefined reference to `esp_ble_gap_set_security_param'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDevice.o): In function `BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)':
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEDevice.cpp:134: undefined reference to `esp_ble_gap_security_rsp'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDevice.o): In function `BLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEDevice.cpp:216: undefined reference to `esp_ble_gatts_register_callback'
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEDevice.cpp:228: undefined reference to `esp_ble_gap_set_security_param'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLECharacteristic.o):(.literal._ZN17BLECharacteristic21handleGATTServerEventE20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x8): undefined reference to `esp_ble_gatts_send_response'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLECharacteristic.o): In function `BLECharacteristic::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*)':
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLECharacteristic.cpp:559: undefined reference to `esp_ble_gatts_send_response'
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLECharacteristic.cpp:559: undefined reference to `esp_ble_gatts_send_response'
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLECharacteristic.cpp:559: undefined reference to `esp_ble_gatts_send_response'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDescriptor.o): In function `BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*)':
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEDescriptor.cpp:184: undefined reference to `esp_ble_gatts_send_response'
/Users/Robert/esp/esp32-ota-https/build/cpp_utils/libcpp_utils.a(BLEDescriptor.o):/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEDescriptor.cpp:213: more undefined references to `esp_ble_gatts_send_response' follow
collect2: error: ld returned 1 exit status
make: *** [/Users/Robert/esp/esp32-ota-https/build/esp32-ota-https.elf] Error 1

nkolban, 2017-12-22:
Many thanks for this post.  This is a good one.  I have never personally tried to compile the BLE C++ classes without all the base ESP32 BLE linked in.  Fortunately, I believe i know how to resolve this puzzle.  At compilation time, we know what settings are enabled/disabled via make menu config so now I will examine this in detail and start work on a resolution.

chegewara, 2017-12-22:
To me it looks like you did compile, then you turn on ble in menuconfig and recompile. But you have to perform `make clean` before you recompile because some libraries were linked before and now they are no longer available because precompile definitions which cut off unused ble code.

robpelu, 2017-12-22:
Hello there, actually and those errors happen after a make clean.
To me is related to disabled elements but I'm willing to test anything you might need to debug the issue.

Thanks !

chegewara, 2017-12-22:
There has to be somewhere included BLE header that suppose not to be. Just check thru your code.

nkolban, 2017-12-22:
The good news is that I was able to recreate and resolve quickly.  There are fourcompilation scenarios:

* GATTC and GATTS enabled
* GATTS only enabled
* GATTC only enabled
* Neither GATTS or GATTC enabled

We can detect all these combinations at compile time.  We now detect if a subsystem is disabled and no longer *register* with it (as it won't be present).  Previously we blindly registered with all the BLE subsystems.  After having made the changes, I compiled the code against the distinct permutations and they now compile cleanly.  I have good confidence that your solution will now compile and run correctly.  Let me ask you to download the very latest release of this library and let's see if we are now good.  If not, post back and we'll pick it up again.

robpelu, 2017-12-22:
Thanks Neil, I've tried but now it's not finding esp_bt.h

CXX build/cpp_utils/BLEAdvertisedDevice.o
CXX build/cpp_utils/BLEAdvertising.o
CXX build/cpp_utils/BLECharacteristic.o
CXX build/cpp_utils/BLECharacteristicMap.o
CXX build/cpp_utils/BLEClient.o
/Users/Robert/esp/esp-idf/components/cpp_utils/./BLEClient.cpp:10:20: fatal error: esp_bt.h: No such file or directory
compilation terminated.
make[1]: *** [BLEClient.o] Error 1
make: *** [component-cpp_utils-build] Error 2
_______________________________________________

Here you could see my menuconfig for BLE.
[image]

robpelu, 2017-12-22:
What's the minimun that I could enable for just a BLE client ?
Do I need Classic Bluetooth and SMP module  ?

nkolban, 2017-12-22:
He he he ... what a mess.  I can explain the "esp_bt.h" puzzle.  About 3 weeks ago, Espressif renamed a header in their ESP-IDF ... they renamed "bt.h" to "esp_bt.h".   This meant that the BLE C++ libraries (this repository) also had to rename the include from bt to esp_bt.h.  My bet is that you have just downloaded the latest BLE C++ source files from this repository but you have **not** refreshed the ESP-IDF (master) from the ESP-IDF repository.

robpelu, 2017-12-22:
Ahhhh, that's exactly what I did.
Ok will refresh the IDF as well.

Thanks for the explanation.

robpelu, 2017-12-23:
OK great, this worked nicely, thanks for the help !

One more thing tough, connection to a BLE device is failing with the following:
I (5762) BLE: Scanning sample starting
I (5762) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (6242) BLE: conectando a ea:ae:85:ec:ef:e4
E (6242) BT: btc_gattc_call_handler()
I (6702) wifi: pm start, type:0

E (36252) BT: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0100
E (36262) BT: p_conn is NULL in bta_gattc_conn_cback

I've been testing in Linux with Gatttool and I have to use the "-t random" switch to make it connect. Do you know why is that and if there is a similar option in the ESP32 ?

Thanks !

chegewara, 2017-12-23:
It some bug introduced with latest esp-idf changes. Ive seen it but didnt report it yet. Its enough if you will restart or reconnect, dont remember.

robpelu, 2017-12-26:
Thanks chegewara, after a couple of scans is connecting. I've read (don't remember where) that if you call connect directly with a MAC address the connection fails, but if you use a scan the struct is filled correctly and that works for me. 
Thanks.

closed 2018-03-30 by robpelu