nkolban/esp32-snippets #309: Able to Pain but unable to connect

ejbommer, 2017-12-22:
I'm currently experimenting with a version of Arduino code (BLE_uart.ino)that says was based off of your C++ work. Using the nRF application I can connect, see the services, and exchange data with the ESP32 without issue. If I try to connect using my android or PC Bluetooth service, my ESP32 name is populated in the discovered items and it pairs, but on my phone it wont connect and on my PC it gives me an "Authenication error". I am assuming there is a service that is not needed by nRF but is required by all other devices. I want to run one of the many Bluetooth terminal programs, but I'm unable since I can not connect. Thanks for any advice.
# **Serial output Monitor after connecting with the nRF:**
 Waiting a client connection to notify...
*** Sent Value: 0 ***
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
*** Sent Value: 5 ***
# **Serial output Monitor after connecting with the Phone Bluetooth:**
Waiting a client connection to notify...
E (32558) BT: lmp_version_below LMP version 6 < 8
E (32560) BT: l2cble_start_conn_update, the last connection update command still pending.
E (33045) BT: FOR LE SC LTK IS USED INSTEAD OF STK
*** Sent Value: 0 ***
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
*** Sent Value: 5 ***
# **Serial output Monitor after connecting with the PC Bluetooth:  No data after Value:0**
Waiting a client connection to notify...
E (20222) BT: lmp_version_below LMP version 6 < 8
*** Sent Value: 0 ***
E (23488) BT: btm_sec_clr_temp_auth_service() - no dev CB
E (23489) BT: Device not found
E (23489) BT: BTA got unregistered event id 31
E (23489) BT: BTA got unregistered event id 31

# **Code:**
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_TX,
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
                      
  pCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_RX,
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {

  if (deviceConnected) {
    Serial.printf("*** Sent Value: %d ***\n", txValue);
    pCharacteristic->setValue(&txValue, 1);
    pCharacteristic->notify();
    txValue++;
  }
  delay(1000);
}

chegewara, 2017-12-22:
Currently security is not implemented in Arduino library yet. As you have rightly noticed iphone and pc are trying to authenticate but thay cant. I cant say whats going on with iPhone, because esp32 is sending new values which means it is connected but with pc connection is just dropped.

ejbommer, 2017-12-22:
Thanks, I was wondering if it was something like that.

If I switch to writing in C++, is this supported? Or is this an issue more with the ESP32?

chegewara, 2017-12-22:
It is supported in esp-idf library version, but there is some issue i have to solve. Unlucky my development laptop just cant see esp32 anymore and i have to solve this first. BLESecurity is on top of my list todo. If you feel comfortable with esp-idf environment then you can try and give me some feedback. Thanks.

chegewara, 2017-12-22:
@ejbommer what application on PC you are using to connect to esp32?

ejbommer, 2017-12-23:
I was using window 7 devises and printers, Add a Bluetooth device.  My ESP32 device will display, but when I click add is when the error happens.

chegewara, 2017-12-23:
Thats normal. Windows will add, pair esp32 and then will disconnect since windows does not know what to do with services and characteristics your esp32 is providing. You need some application that will connect and communicate with those.

ejbommer, 2017-12-23:
It's funny I never done any wireless communication before.    Then I purchased an esp 8266 and found wifi was pretty easy to get working, thanks to all the work people like you do.  I thought Bluetooth was going to be the same, I was wrong.
Thanks for the information.  I'm not currently trying to get my PC to connect to the esp32, I only tried it becaus my phone failed so I wondered if the PC would act any  different.

chegewara, 2017-12-23:
```
Waiting a client connection to notify...
E (32558) BT: lmp_version_below LMP version 6 < 8
E (32560) BT: l2cble_start_conn_update, the last connection update command still pending.
E (33045) BT: FOR LE SC LTK IS USED INSTEAD OF STK
*** Sent Value: 0 ***
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
*** Sent Value: 5 ***
```

This logs you provided shows that esp32 or your phone is requesting encrypted connection and key got exchanged, then esp32 is sending some values which means there is connection between those two.

ejbommer, 2017-12-23:
That's what I was wondering.  The print out only occurs if it thinks it's connected, but when I open up my Bluetooth devices on my phone it just shows it paired, but not connected.  If I click connect on my phone, which Grey's out once the connection is made.  Its not greyed out and I  see no signs of any activity when I press it.

chegewara, 2017-12-23:
Like i said, we didnt implemented security yet. This is what you can do just for now for tests. Add this code in setup after Device::init():

 ```
   esp_ble_auth_req_t auth_req = ESP_LE_AUTH_NO_BOND;     
    esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;          
    uint8_t key_size = 16;     
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
```
I didnt check it by myself, but should work.

ejbommer, 2017-12-23:
## I have added your code suggestions
void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

###   //  Test Code Start

    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_NO_BOND;     
    esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;          
    uint8_t key_size = 16;     
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

###   //Test Code End

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
## This is my new output:
Waiting a client connection to notify...
E (56194) BT: Non bonding: No keys will be exchanged
E (56196) BT: l2cble_start_conn_update, the last connection update command still pending.
*** Sent Value: 0 ***
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
*** Sent Value: 5 ***

chegewara, 2017-12-23:
This is too few logs, there is something else before:
```
Waiting a client connection to notify...
E (56194) BT: Non bonding: No keys will be exchanged
E (56196) BT: l2cble_start_conn_update, the last connection update command still pending.
```

ejbommer, 2017-12-23:
This everything on the screen after I press the reset button.  I only see the 2 errors after I try to connect.
## *************************************************
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
Waiting a client connection to notify...
*** Sent Value: 0 ***
E (23804) BT: Non bonding: No keys will be exchanged
E (23805) BT: l2cble_start_conn_update, the last connection update command still pending.
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
*** Sent Value: 5 ***

chegewara, 2017-12-23:
Ok, im going to ask to change logging level to verbose.
`Tools->Core debug level->Verbose`

ejbommer, 2017-12-23:
I'm not seeing that option in Arduino 1.8.5

chegewara, 2017-12-23:
[image]

ejbommer, 2017-12-23:
You must have something enabled that I don't.  Are you running linux or some other then windows?

[image]

ejbommer, 2017-12-23:
I change changed my board to Dev Board, now I have it.  I will get that setting a try.

ejbommer, 2017-12-23:
Yep that worked

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
[D][BLEDevice.cpp:58] createServer(): >> createServer
[D][BLEServer.cpp:297] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1755] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:301] registerApp(): << registerApp
[D][BLEDevice.cpp:61] createServer(): << createServer
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:76] createService(): >> createService - 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1680] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, inst_id: 0]]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:160] setHandle(): >> setHandle - Handle=0x28, service UUID=6e400001-b5a3-f393-e0a9-e50e24dcca9e)
[D][BLEService.cpp:166] setHandle(): << setHandle
[D][BLEService.cpp:89] executeCreate(): << executeCreate
[D][BLEServer.cpp:93] createService(): << createService
[D][BLEService.cpp:188] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:191] addCharacteristic(): Adding characteristic: uuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEService.cpp:203] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:71] addDescriptor(): >> addDescriptor(): Adding UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff to UUID: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, handle: 0xffff Notify 
[D][BLECharacteristic.cpp:73] addDescriptor(): << addDescriptor()
[D][BLEService.cpp:188] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:191] addCharacteristic(): Adding characteristic: uuid=6e400002-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
[D][BLEService.cpp:203] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:557] setCallbacks(): >> setCallbacks: 0x3ffdb4c4
[D][BLECharacteristic.cpp:559] setCallbacks(): << setCallbacks
[D][BLEService.cpp:125] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
[D][BLECharacteristic.cpp:82] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:93] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1633] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:574] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e
[D][BLECharacteristic.cpp:576] setHandle(): << setHandle
[D][BLEDescriptor.cpp:63] executeCreate(): >> executeCreate(): UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_DESCR_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
[D][BLEUtils.cpp:1621] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 43 0x2b, service_handle: 40 0x28, char_uuid: 00002902-0000-1000-8000-00805f9b34fb]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
[D][BLEDescriptor.cpp:275] setHandle(): >> setHandle(0x2b): Setting descriptor handle to be 0x2b
[D][BLEDescriptor.cpp:277] setHandle(): << setHandle()
[D][BLEDescriptor.cpp:87] executeCreate(): << executeCreate
[D][BLECharacteristic.cpp:134] executeCreate(): << executeCreate
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:82] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:93] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_EVT
[... cut ...]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1261] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!

ejbommer, 2017-12-23:
Very cool feature that for teaching me about it.

chegewara, 2017-12-23:
Your phone is connected but does not request for notifications:
`[D][BLECharacteristic.cpp:509] notify(): << notifications disabled; ignoring`
In nRF you have to click 3 arrows to request for notifications. On your phone your client app has to connect to esp32, find characteristic and request notifications.

ejbommer, 2017-12-23:
Do you think the Android operating system used to add Bluetooth devices is not replying to the notification is because the security features are not enabled?

chegewara, 2017-12-23:
Yes, its exactly this. Android operating system does not know what service and characteristic need to ask for notifications, except there is some specialized service and characteristic that anrdoid os can use, but i dont know any.
You can try to use BLE_uart with nRF UART, that should have work.

rweller59, 2017-12-31:
Can you provide a bit more explanation on this issue. Using espidf + cpp_utils. Everything upgraded yesterday. Esp is ble server. I have not set any security properties. I can connect but can no longer read services or characteristics (testing with nRF). I saw the following message come up in the console just once - every other time the esp acts as if connection is fine and notifications are sent as normal.

I (175319) BLEDevice: ESP_GAP_BLE_SEC_REQ_EVT
E (175319) BT: lmp_version_below LMP version 6 < 8
E (175322) BT: l2cble_start_conn_update, the last connection update command still pending.
E (175680) BT: FOR LE SC LTK IS USED INSTEAD OF STK
I (175714) BLEDevice: key type = ESP_LE_KEY_LENC
I (175714) BLEDevice: key type = ESP_LE_KEY_PENC
I (175715) BLEDevice: key type = ESP_LE_KEY_LID
I (175719) BLEDevice: key type = ESP_LE_KEY_LCSRK
I (175741) BLEDevice: key type = ESP_LE_KEY_PID
I (175877) BLEDevice: key type = ESP_LE_KEY_PCSRK

chegewara, 2017-12-31:
Explanation:
`I (175319) BLEDevice: ESP_GAP_BLE_SEC_REQ_EVT`
issued secure connection request from peer device
`E (175319) BT: lmp_version_below LMP version 6 < 8`
this is not error, thats how devs logs some messages, its just debug info
`E (175322) BT: l2cble_start_conn_update, the last connection update command still pending.`
ive seen this message lately, it just means that you are requesting secure connection but one is already requested and still pending, you can remove from your code BLEDevice::setEncryptionLevel() or just ignore (my advice - ignore)
`E (175680) BT: FOR LE SC LTK IS USED INSTEAD OF STK`
again, its just info that LE secure connection LTK(long term key) will be used instead of STK(short term key)
```
I (175714) BLEDevice: key type = ESP_LE_KEY_LENC
I (175714) BLEDevice: key type = ESP_LE_KEY_PENC
I (175715) BLEDevice: key type = ESP_LE_KEY_LID
I (175719) BLEDevice: key type = ESP_LE_KEY_LCSRK
I (175741) BLEDevice: key type = ESP_LE_KEY_PID
I (175877) BLEDevice: key type = ESP_LE_KEY_PCSRK
```
here you can see what kind key types are exchanged between devices
Now, when we have already implemented security layer, you dont have to setup any security to use secure encrypted connection. Device that you are connecting to is sending request to use secure connection and by default its setup to accept those requests.

About reading characteristics, there should not be issue, but i will double check.

rweller59, 2017-12-31:
I used nRF to delete bonded information. After that, both nRF and my app were both able to communicate as previously, getting service and characteristic information.

So, this is a one-time issue because of security upgrade to espdif/cpp_utils? Sounds like unless the client  requests a secure connection, I shouldn't have this type of issue again?

chegewara, 2017-12-31:
You will have this issue every time when you have bound devices and you will reflash esp32. you need to unbind from nRF every time you reflash esp32. Its natural because you are trying to use old keys with nRF and new with esp32

rweller59, 2017-12-31:
ok. thanks. and yes that does make sense. 
Do you know offhand if there is a standard ble method to call (on the client side) to indicate a security key mismatch? If not, I'll do some research.

chegewara, 2017-12-31:
If there is key mismatch you should have not be able to connect. If you will study cpp_utils examples from security folder, you can see there is BLESecurityCallback. In that callback is onAuthenticationComplete() called on AUTH_CMPL_EVT with parameter esp_ble_auth_cmpl_t.
This is structure of esp_ble_auth_cmpl_t and as you can see you can check if authentication is completed with success and if not what is the failure reason (code):
```
{
    esp_bd_addr_t         bd_addr;               /*!< BD address peer device. */
    bool                  key_present;           /*!< Valid link key value in key element */
    esp_link_key          key;                   /*!< Link key associated with peer device. */
    uint8_t               key_type;              /*!< The type of Link Key */
    bool                  success;               /*!< TRUE of authentication succeeded, FALSE if failed. */
    uint8_t               fail_reason;           /*!< The HCI reason/error code for when success=FALSE */
    esp_ble_addr_type_t   addr_type;             /*!< Peer device address type */
    esp_bt_dev_type_t     dev_type;              /*!< Device type */
} esp_ble_auth_cmpl_t;  
```

rweller59, 2017-12-31:
Before I cleared the bonded info (using nRF), I was able to connect and nRF was showing generic service and characteristic uuid's. The ESP server thought that it had a valid connection and continued to send values until I disconnected the client. 
I can probably do some testing there but probably can't get to it until tomorrow.

chegewara, 2017-12-31:
There is a lot more than that. Maybe keys are valid when you reflash esp32 because android has copy of it and esp32 has one set keys per device. I cant say much about how keys are created. This is how i work with secure connections and any connections that are using bonding. Each time i reflash esp32 im deleting bonding on android phone or windows laptop and im pairing devices again. Sometimes before i reflash esp32 i erase_flash to delete all whitelist info from NVS, because even if you make re-flash all bound info are kept in NVS.

BTW security was introduced in esp-idf some time ago, just we add it here, into cpp_utils recently and it still may be a little bit bugged

chegewara, 2018-01-05:
Are we clear in this matter? Can we close this issue?

ejbommer, 2018-01-05:
I'm good thank you for all your help.

hansmbakker, 2018-12-20:
@ejbommer can you then close this issue to keep the issue list clean?

cc @chegewara

closed 2018-12-20 by ejbommer

pana216, 2019-08-25:
> Your phone is connected but does not request for notifications:
**_> [D][BLECharacteristic.cpp:509] notify(): << notifications disabled; ignoring_**
> In nRF you have to click 3 arrows to request for notifications. On your phone your client app has to connect to esp32, find characteristic and request notifications.

Does anyone know how we can stop the ESP32 advertising the notification if the client explicitly doesn't want to get notified anymore? I mean why is the ESP32 ignoring the disabled notification?

chegewara, 2019-08-26:
@pana216 https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L512-L514
Which means is not sending notification. The problem is when there is more than 1 client connected and at least 1 is registered to get notifications, then all client will get notifications.