nkolban/esp32-snippets #314: Design and implement a BLE2904 descriptor helper

nkolban, 2017-12-23:
The BLE descriptor with UUID 0x2904 has been asked to be modeled as a class for ease of use.  Since this is a BLE architected descriptor it does indeed make sense to model this.

The BLE descriptor 0x2904 can be found documented here:

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml

and is also known as the "Characteristic Presentation Format".

nkolban, 2017-12-23:
First pass at code changes  made.

Example of usage in a BLE Server:

```
BLE2904 *p2904Descriptor = new BLE2904();
p2904Descriptor->setFormat(BLE2904::FORMAT_UTF8);
pCharacteristic->addDescriptor(p2904Descriptor);
```

chegewara, 2017-12-24:
Units parameter in this descriptor is described here:
https://www.bluetooth.com/specifications/assigned-numbers/units

axa88, 2019-08-13:
Curious if this works at all as I find adding a BLE2904 to my characteristic is recognized as a 0x2902 CCCD in both nRF Connect and LightBlue dev apps.

Furthermore adding it to a characteristic that already has a BLE2902 descriptor added, screws up the the 2902... so im not really sure how to use the BLE2904 if at all.

Any advise?

chegewara, 2019-08-14:
Hi @axa88 
i just tested this code and with nRF connect i can see both, 2902 and 2904 descriptors:
```
		BLE2902* p2902Descriptor = new BLE2902();
		BLE2904* p2904Descriptor = new BLE2904();
		pCharacteristic->addDescriptor(p2902Descriptor);
		pCharacteristic->addDescriptor(p2904Descriptor);
```

axa88, 2019-08-14:
thanks to look into this.
Trying again, albeit on a different phone running nRF app, i find your right this attribute does work and i now get only one of each type of descriptor.

closed 2019-08-14 by chegewara

axa88, 2019-08-14:
Oddly enough this seems to be a bug in either the Samsung S7 running Android 8.0.0 level 26 Oreo supporting Bluetooth v4.2
When the esp32 sets a service with characteristics having a x2904 descriptor, it is unrecognized:

[image]

While a HTC Bolt running Android 7.1 level 25 Nuaget supporting Bluetooth 4.1 connected to the same esp32 at the same time displays the x2904 descriptor:

[image]

As I found this issue using more than on app, I am estimating it is not any particular application.

chegewara, 2019-08-14:
Maybe you have bonded esp32 on S7 and you need to refresh GATT attributes?
Android remember old GATT attributes and even if you change GUUIDs completely it will display old values.

Next to the `disconnect` you have context menu, there is `refresh services`.

axa88, 2019-08-14:
Wow chegewara you have acquired some great knowledge about Bluetooth and I really appreciate what you guys have done with support here.

That was the issue.  I suppose I will learn more about this when I write the Android peripheral side.