nkolban/esp32-snippets #316: We could have issue in RemoteCharacteristic->writeValue()

chegewara, 2017-12-23:
Probably we have issue in RemoteCharacteristic->writeValue(). When i run SampleClient for long enough i found that BT stack is trimming longer messages message to 20 bytes. Example code that cause:
 ```
stringStream << "Time since boot: " << tv.tv_sec;
pRemoteCharacteristic->writeValue(stringStream.str());
```
and logs:

```
D (17262779) BLERemoteCharacteristic: >> writeValue(), length: 22
D (17262779) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x3ffce1d8), owner: <N/A> for writeValue
D (17262788) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x3ffce1d8), owner: writeValue
E (17262796) BT: attribute value too long, to be truncated to 20
V (17262797) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x3ffce1d8), owner: writeValue for writeValue
D (17262804) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (17262822) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (17262827) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (17262834) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x3ffce1d8), owner: writeValue
V (17262843) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x3ffce1d8), owner: <N/A>
D (17262852) BLERemoteCharacteristic: << writeValue
D (17263857) SampleClient: Setting the new value
D (17263857) BLERemoteCharacteristic: >> writeValue(), length: 22
D (17263857) FreeRTOS: Semaphore taking: name: WriteCharEvt (0x3ffce1d8), owner: <N/A> for writeValue
D (17263866) FreeRTOS: Semaphore taken:  name: WriteCharEvt (0x3ffce1d8), owner: writeValue
E (17263874) BT: attribute value too long, to be truncated to 20
V (17263875) FreeRTOS: >> wait: Semaphore waiting: name: WriteCharEvt (0x3ffce1d8), owner: writeValue for writeValue
D (17263882) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
D (17263900) BLEUtils: GATT Event: ESP_GATTC_WRITE_CHAR_EVT
D (17263905) BLEUtils: [status: ESP_GATT_OK, conn_id: 0, handle: 42 0x2a]
V (17263912) FreeRTOS: Semaphore giving: name: WriteCharEvt (0x3ffce1d8), owner: writeValue
V (17263921) FreeRTOS: << wait: Semaphore released: name: WriteCharEvt (0x3ffce1d8), owner: <N/A>
D (17263930) BLERemoteCharacteristic: << writeValue
D (17264935) SampleClient: Setting the new value
```

nkolban, 2017-12-25:
Recreated trivially ... examining in more depth now.

chegewara, 2017-12-25:
It may be or not the issue in esp-idf BT stack, since error message is from BT, also message is truncated by esp-idf to 20 but MTU is 23, or maybe its issue with write procedure when PDU is too big and we should send 20 bytes not 22 as is now.

nkolban, 2017-12-25:
I tried running nRFConnect in its server mode.  Using this, we can tweak the MTU accepted by the server.  The default appears to be 23 bytes and there appears to be a 3 byte overhead for the write request.  If I sent 40 bytes, I got the error.  If I increased the MTU to be 43, the problem was no longer reported and all was well.

chegewara, 2018-01-01:
Possibly it has been fixed. Closing for now.

closed 2018-01-01 by chegewara