nkolban/esp32-snippets #318: Compiler Option Enable c++ exceptions

thiamugb, 2017-12-26:
I have a problem when i want to make my project i have this error and i see nothing in the menuconfig help.
Here is my error!!
**In file included from /Users/thiammac/Documents/workspace/cpp3/components/cpp_utils/./BLEClient.h:18:0,
                 from /Users/thiammac/Documents/workspace/cpp3/components/cpp_utils/./BLEScan.h:16,
                 from /Users/thiammac/Documents/workspace/cpp3/components/cpp_utils/./BLEAdvertisedDevice.h:17,
                 from /Users/thiammac/Documents/workspace/cpp3/components/cpp_utils/./BLEAdvertisedDevice.cpp:18:
/Users/thiammac/Documents/workspace/cpp3/components/cpp_utils/./BLEExceptions.h:13:2: error: #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."
 #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."
  ^
make[1]: *** [BLEAdvertisedDevice.o] Error 1
make: *** [component-cpp_utils-build] Error 2**
Best regards

chegewara, 2017-12-26:
Is it arduino-ide or esp-idf?

thiamugb, 2017-12-26:
Its esp-idf i use eclipse in c++

Cordialement 
Thiam babacar
Contact :+1(514)-951-1353

chegewara, 2017-12-26:
https://github.com/nkolban/esp32-snippets/issues/231#issuecomment-347039923

thiamugb, 2017-12-26:
> #231 (comment)

So i see them it’s different!!
In my menuconfig i haven’t c++exception in compiler options

chegewara, 2017-12-26:
Then you have to update esp-idf

thiamugb, 2017-12-26:
i update then and after i have an error with this code saying:
 error: 'class BLEServer' has no member named 'setEncryptionLevel'
   pServer->setEncryptionLevel(ESP_BLE_SEC_ENCRYPT_NO_MITM);
            ^

chegewara, 2017-12-26:
Yes, you have old example. I redesigned lately securit library. Check #313

thiamugb, 2017-12-26:
Yes i see the error is the new security option.
So what do you want edit file in cpp_utils/BLEclient and server using #313 or delete example test https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleSecureServer.cpp or client.
I am confused!!!!

chegewara, 2017-12-26:
In #313 i redesigned security but for some reason i didnt delete obsolete files. All new example files are in security subfolder.

thiamugb, 2017-12-26:
Okay i see.
Me i thought we used just BLETests folder. All is done my test is correct.
thanx see u ... 
When i used server BLE its not easy to see BLE in another periphery ( for example my cellphone) because if i used micro python my BLE is found easily.
Its an configuration or No ,

chegewara, 2017-12-28:
I dont know what is the issue, i can find anything i flash onto my two esp32-DevKitC devices with no problem. Im using to tests samsung SGS4 and laptop with windows 10.

BladeRunner68, 2018-01-06:
I'm having major issues with this when trying to compile on Platformio - was there a cure at all?

arnopoulos, 2018-02-12:
@BladeRunner68 I am also running into the same issue. I was able to wrap the definition and uses of `BLEDisconnectedException` and `BLEUuidNotFoundException` inside of `#if CONFIG_CXX_EXCEPTIONS == 1` blocks. However I hit a roadblock with uses of `esp_bt.h`. I don't know enough about `esp_bt.h` to fix this issue. Which is unfortunate because I need to use `BLEAdvertisementData` for a project I'm working on and that was introduced in `v0.4.8` the same version that introduces the compilation error.

arnopoulos, 2018-03-24:
@BladeRunner68 I was able to fix this issue by upgrading the platform inside of the PlatformIO Home page. I'm not sure if this will fix the issue for you but it definitely fixed the issue for me.

chegewara, 2018-07-23:
How about updating arduino environment? Its been enabled long time ago.

MikeyMoMo, 2018-07-23:
On 1.8.4.  Isn't that new enough?

chegewara, 2018-07-23:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code

closed 2018-11-13 by chegewara