nkolban/esp32-snippets #319: BLECharacteristic ESP_GATTS_CONF_EVT semaphore issue

chegewara, 2017-12-26:
Maybe its issue with my application, but im sending only one notification in one characteristic and im getting bunch of semaphore.give() calls and just semaphore.take() call.

https://github.com/nkolban/esp32-snippets/blob/f28769c8d2e7fd313ed78644d477fcd918c42203/cpp_utils/BLECharacteristic.cpp#L415-L418
```
D (41762) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
V (41768) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdfad8), owner: <N/A>
V (41776) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdfc94), owner: <N/A>
V (41783) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdffa0), owner: <N/A>
V (41790) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0570), owner: <N/A>
V (41797) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe075c), owner: <N/A>
V (41805) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0948), owner: notify
V (41812) FreeRTOS: << wait: Semaphore released: name: ConfEvt (0x3ffe0948), owner: <N/A>
V (41812) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0b7c), owner: <N/A>
D (41820) BLECharacteristic: << notify
V (41828) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0db0), owner: <N/A>
V (41839) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0fe4), owner: <N/A>
V (41846) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe1218), owner: <N/A>
V (41853) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe144c), owner: <N/A>
D (41861) BLEServer: << handleGATTServerEvent
D (42081) BLECharacteristic: >> setValue: length=5, data=00001050ff, characteristic UUID=00002a4d-0000-1000-8000-00805f9b34fb
D (42082) BLECharacteristic: << setValue
D (42086) BLECharacteristic: >> notify: length: 5
D (42090) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (42099) GeneralUtils: 0000 00 00 10 50 ff                                   ...P.
D (42107) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffe0948), owner: <N/A> for notify
D (42115) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffe0948), owner: notify
V (42123) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffe0948), owner: notify for notify
D (42124) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CONF_EVT
D (42139) BLEUtils: GATT ServerEvent: ESP_GATTS_CONF_EVT
D (42145) BLEUtils: [status: ESP_GATT_OK, conn_id: 0x00]
D (42150) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
V (42156) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdfad8), owner: <N/A>
V (42164) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdfc94), owner: <N/A>
V (42171) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffdffa0), owner: <N/A>
V (42178) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0570), owner: <N/A>
V (42185) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe075c), owner: <N/A>
V (42193) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0948), owner: notify
V (42200) FreeRTOS: << wait: Semaphore released: name: ConfEvt (0x3ffe0948), owner: <N/A>
V (42200) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0b7c), owner: <N/A>
D (42208) BLECharacteristic: << notify
V (42216) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0db0), owner: <N/A>
V (42227) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe0fe4), owner: <N/A>
V (42234) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe1218), owner: <N/A>
V (42241) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe144c), owner: <N/A>
D (42249) BLEServer: << handleGATTServerEvent

```

nkolban, 2017-12-26:
That code and the code after it looks pretty sad I'm afraid.  The model is that we have event handlers that get invoked when an event is received.  Since we need event handles for all the characteristics, we keep a tree/chain of event handlers.  When an event arrives, we give each handler the opportunity to play be given it to do something.  Unfortunately, the handler you point to and some of the ones following it simply "assume" that the event is meant for them and don't ask "Hey ... here is an event ... is this event for ME?" ... it is like a letter arriving at a shared residence and everyone opening it and reading it.  Instead, only the person to whom it is addressed should open it.

chegewara, 2017-12-26:
Sorry, i didnt want to :(

chegewara, 2017-12-26:
This part also looks suspicious:
https://github.com/nkolban/esp32-snippets/blob/f28769c8d2e7fd313ed78644d477fcd918c42203/cpp_utils/BLECharacteristic.cpp#L425-L427

chegewara, 2017-12-26:
https://github.com/nkolban/esp32-snippets/blob/f28769c8d2e7fd313ed78644d477fcd918c42203/cpp_utils/BLECharacteristic.cpp#L422

nkolban, 2018-01-04:
I had a study of this code and made some small changes that didn't change function (added a few more diagnostics and cleaned up some formatting).  I have convinced myself that the even though we are seeing more semaphore give() calls than take() calls .... this is ok.

Here is the thinking.

A take() request says "Take the semaphore and if it is not present, wait for it to be present and then take it"

A give() requests says "If the semaphore is not present, set it to be present".

Notice what give() doesn't say ... it doesn't say what should happen if the semaphore is ALREADY present.  The answer is that it does nothing.  The effect after performing a give() is that the semaphore is present.  If it wasn't present, then it is now ... however if it was present ... it is STILL present afterwards.

We loosely think of a semaphore as being returned by the person who took it, but that needn't be the case.  I'm going to suggest that this is not an issue at this time but we should keep our eye on it in case it has some other side effect we care about.

chegewara, 2018-01-04:
We will see, but i have another simpthom that can be related to semaphores. Im having server now (hid device), which is paired with windows 10 laptop. All seems to work, but when i reset they cant reconnect even if windows see esp32 as paired. I will study it more, because it can be related to bonding, and esp32 does not store information about bonding with laptop. This is strange log, as we can see we have bunch of advertising tasks. Its because laptop is trying to connect and disconnecting very quick, and we have advertising->start() every time onDisconnect() event:

```
D (88660) BLEServer: BLEServer ... handling GAP event!
D (90880) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (90880) BLEUtils: [status: 0]
D (90880) BLEServer: BLEServer ... handling GAP event!
D (91590) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (91590) BLEUtils: [status: 0]
D (91590) BLEServer: BLEServer ... handling GAP event!
D (92220) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (92220) BLEUtils: [status: 0]
D (92220) BLEServer: BLEServer ... handling GAP event!
D (92840) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (92840) BLEUtils: [status: 0]
D (92840) BLEServer: BLEServer ... handling GAP event!
D (93470) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (93470) BLEUtils: [status: 0]
D (93470) BLEServer: BLEServer ... handling GAP event!
D (94850) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (94850) BLEUtils: [status: 0]
D (94850) BLEServer: BLEServer ... handling GAP event!
D (95500) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (95500) BLEUtils: [status: 0]
D (95500) BLEServer: BLEServer ... handling GAP event!
D (96160) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (96160) BLEUtils: [status: 0]
D (96160) BLEServer: BLEServer ... handling GAP event!
D (96790) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (96790) BLEUtils: [status: 0]
D (96790) BLEServer: BLEServer ... handling GAP event!
D (97390) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (97390) BLEUtils: [status: 0]
D (97390) BLEServer: BLEServer ... handling GAP event!
D (98190) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (98190) BLEUtils: [status: 0]
D (98190) BLEServer: BLEServer ... handling GAP event!
D (99010) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (99010) BLEUtils: [status: 0]
D (99010) BLEServer: BLEServer ... handling GAP event!
D (99640) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (99640) BLEUtils: [status: 0]
D (99640) BLEServer: BLEServer ... handling GAP event!
D (100280) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (100280) BLEUtils: [status: 0]
```

closed 2018-10-04 by chegewara