nkolban/esp32-snippets #320: BLE pycom is not visible on a bLEscanner

thiamugb, 2017-12-27:
when I ran SampleServerEncryption to use my ESP32 as a server nothing is displayed if I do a scan with my tel using a BLEscanner And i can stop monitor with ctrl + c .
I use eclipse c++ on mac os x  and terminal to display monitor.
Need Help

chegewara, 2017-12-27:
Could you paste logs from esp32?

thiamugb, 2017-12-27:
> le: 0x28
D (915) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffd23b0), owner: <N/A> for executeCreate
D (923) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffd23b0), owner: executeCreate
E (931) BT: osi_mem_dbg_record invalid !!

D (935) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_EVT
D (942) BLEUtils: GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (948) BLEUtils: [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: 0eea3ca1-782b-47b9-bbcd-dbbddf10a38e]
D (959) BLEServer: >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (965) BLECharacteristic: >> setHandle: handle=0x2a, characteristic uuid=0eea3ca1-782b-47b9-bbcd-dbbddf10a38e
D (975) BLECharacteristic: << setHandle
V (979) FreeRTOS: Semaphore giving: name: CreateEvt (0x3ffd23b0), owner: executeCreate
D (986) BLEServer: << handleGATTServerEvent
V (990) FreeRTOS: >> wait: Semaphore waiting: name: CreateEvt (0x3ffd23b0), owner: <N/A> for executeCreate
V (1000) FreeRTOS: << wait: Semaphore released: name: CreateEvt (0x3ffd23b0), owner: executeCreate
D (1008) BLEDescriptor: >> executeCreate(): UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff
D (1017) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffd24fc), owner: <N/A> for executeCreate
D (1026) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffd24fc), owner: executeCreate
D (1034) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_DESCR_EVT
D (1042) BLEUtils: GATT ServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
D (1048) BLEUtils: [status: ESP_GATT_OK, attr_handle: 43 0x2b, service_handle: 40 0x28, char_uuid: 00002902-0000-1000-8000-00805f9b34fb]
D (1060) BLEServer: >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
D (1066) BLEDescriptor: >> setHandle(0x2b): Setting descriptor handle to be 0x2b
D (1073) BLEDescriptor: << setHandle()
V (1077) FreeRTOS: Semaphore giving: name: CreateEvt (0x3ffd24fc), owner: executeCreate
D (1085) BLEServer: << handleGATTServerEvent
V (1089) FreeRTOS: >> wait: Semaphore waiting: name: CreateEvt (0x3ffd24fc), owner: <N/A> for executeCreate
V (1098) FreeRTOS: << wait: Semaphore released: name: CreateEvt (0x3ffd24fc), owner: executeCreate
D (1107) BLEDescriptor: << executeCreate
D (1110) BLECharacteristic: << executeCreate
D (1114) FreeRTOS: Semaphore taking: name: StartEvt (0x3ffd1ff8), owner: <N/A> for start
D (1122) FreeRTOS: Semaphore taken:  name: StartEvt (0x3ffd1ff8), owner: start
D (1130) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_START_EVT
D (1136) BLEUtils: GATT ServerEvent: ESP_GATTS_START_EVT
D (1141) BLEUtils: [status: ESP_GATT_OK, service_handle: 0x28]
D (1147) BLEServer: >> handleGATTServerEvent: ESP_GATTS_START_EVT
V (1153) FreeRTOS: Semaphore giving: name: StartEvt (0x3ffd1ff8), owner: start
D (1160) BLEServer: << handleGATTServerEvent
V (1164) FreeRTOS: >> wait: Semaphore waiting: name: StartEvt (0x3ffd1ff8), owner: <N/A> for start
V (1172) FreeRTOS: << wait: Semaphore released: name: StartEvt (0x3ffd1ff8), owner: start
D (1180) BLEService: << start()
D (1183) SampleServer: Advertising started!
D (1187) BLEAdvertising: >> start: customAdvData: 0, customScanResponseData: 0
D (1194) BLEAdvertising: - advertising service: 9845f7dc-4468-491a-9d8a-d3aba0eb8287
D (1204) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
D (1209) BLEUtils: [status: 0]
D (1211) BLEServer: BLEServer ... handling GAP event!
D (1217) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
D (1223) BLEUtils: [status: 0]
D (1226) BLEServer: BLEServer ... handling GAP event!
D (1233) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (1238) BLEUtils: [status: 0]
D (1240) BLEServer: BLEServer ... handling GAP event!
D (1245) BLEAdvertising: << start
D (1248) SampleServer: Advertising started!

>

chegewara, 2017-12-27:
From logs everything seems to be fine. Advertising starts here. Can you try with different application on you phone or maybe with laptop or something?
```
D (1194) BLEAdvertising: - advertising service: 9845f7dc-4468-491a-9d8a-d3aba0eb8287
D (1204) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
D (1209) BLEUtils: [status: 0]
D (1211) BLEServer: BLEServer ... handling GAP event!
D (1217) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
D (1223) BLEUtils: [status: 0]
D (1226) BLEServer: BLEServer ... handling GAP event!
D (1233) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (1238) BLEUtils: [status: 0]
D (1240) BLEServer: BLEServer ... handling GAP event!
D (1245) BLEAdvertising: << start
D (1248) SampleServer: Advertising started!
```

thiamugb, 2017-12-27:
Yeah i see all is correct.
I try with the application of Nordit on my 2 phones (apple and android) nothing is found.
And my problem is also the monitor does not close if I want to close it with cmd + c or other as if it is planted that I disconnect my esp32 if I want to close.

chegewara, 2017-12-27:
To see all monitor option use:
cmd + T and cmd + H (it works on linux)

About SampleServer_Encrypted, i cant help you with that. Just checked and everything seems works fine. Compiled, installed and found with laptop and android phone (nRF connect).

thiamugb, 2017-12-27:
But I want to know if there is an application other than nrf that we can have on pc ?
Cause if i use micropython create an server with BLE its found with nrf nordic and i can see caracterist and others.

chegewara, 2017-12-27:
Like i said, i can see SampleServer_Encrypted and any other server examples with my windows 10 laptop. I dont have mac OS so i cant confirm or deny its working, but its a standard so it should have works.

thiamugb, 2017-12-27:
of course i will try to see on my windows 10 because I had problems with my esp32 drivers.
But I will try to see for the configuration for the drivers under esp32 thanks again for everything.
Best Regards

thiamugb, 2017-12-27:
No, i say that i have problem to use MSYS2 MINGW32 under windows I could not browse to finish the installation that's why I abandoned windows.