nkolban/esp32-snippets #321: latest IDF update has I2C changes that cause compile errors with cpp_utils I2C.cpp

yaoyogi, 2017-12-27:
Looks like some IDF changes related to the driver API's: i2c_master_read(..) and i2c_master_read_byte(..) using i2c_ack_type_t

squonk11, 2017-12-27:
I am now having the same problem.

yaoyogi, 2017-12-27:
The fix is simple if you need it. Currently the Kolban code relies on a bool value for defining the ack.   The IDF API was updated to use i2c_ack_type_t vs. an int.  So while bool would convert to int before in C, it wont convert to i2c_ack_type_t in C++, hence the compile errors.
  
bool true   ==1   ---> i2c_ack_type_t::I2C_MASTER_NACK
bool false == 0  ---> i2c_ack_type_t::I2C_MASTER_ACK  

In the 2 API's where this broke, the IDF I2C driver calls are taking !ack as the last arg.  Basically for these two cases we could do the following to swap the mapping:

i2c_ack_type_t akker = (!ack) ? i2c_ack_type_t::I2C_MASTER_NACK : i2c_ack_type_t::I2C_MASTER_ACK ;

And where the !ack was being passed, just pass akker.  This should work. 

 I am not using the I2C portion of cpp_utils, but had a similar issue in my I2C layer with this IDF update.  I think Kolban stays close to the IDF updates but I don't know when he decides to update/publish his code so I don't want to make any changes.

squonk11, 2017-12-27:
thankx, I also temporarily use your fix.

Stormhand, 2017-12-27:
I temp fixed it too but now there is another error:

> E (628110) i2c: /home/strm/esp/esp-idf/components/driver/./i2c.c:1008 (i2c_master_read):i2c data read length error
> E (628110) I2C: i2c_master_read: rc=258 Invalid argument
>

nkolban, 2017-12-27:
Sorry guys ... I'll re-test and attend to this tonight.  When 3.0 ESP-IDF formally ships, we will create a "release" of the cpp_utils so that we will have a version that will be locked down (and maintained) against 3.0 and a master that will keep in synch with code base changes post 3.0.  3.0 is a breaking change to cpp_utils and until 3.0 ships, we will be playing catch-up to maintain compatibility.

nkolban, 2017-12-28:
Howdy guys, the error due to the `i2c_ack_type_t` is now resolved and the library compiles cleanly.  Many thanks for the report.  As to the "rc=258" issue ...  we'll need some diagnostics to comment on that one.

yaoyogi, 2018-01-01:
The compile issue has been fixed for awhile.  Perhaps this issue should be closed and if there is other I2C runtime issues a new issue would be easier to track?

nkolban, 2018-01-01:
Howdy my friend, the owner of the issue (yourself) can close this issue anytime you wish.  Anyone else on this issue is just "piggy backing" so don't worry too much about that.  I'm of the opinion that separate and unrelated issues are the ideal way to track.

We can always re-open an issue should the problem warrant.  Every few weeks I walk through the back-issues and ones that are open and awaiting raiser response  who has not responded in at least 14 days ... then the issue is closed with the message that it can be re-opened at any time.

yaoyogi, 2018-01-01:
Issue resolved.  Thank you for quick fix.

closed 2018-01-01 by yaoyogi

yaoyogi, 2018-01-09:
In case anyone was still having problems with the I2C read length error.  It was a bug introduced in IDF I2C.cpp https://github.com/espressif/esp-idf/commit/5fb16d9b2ba756f330249e75968a3110b4a45075 and not Kolban I2C.
The current IDF pull seems to have reversed this mistake, and my I2C layer is working correctly as before.  I'm pretty sure this was the read error another commenter had run into.