nkolban/esp32-snippets #322: HttpServer stuck on request

Stormhand, 2017-12-27:
Im using the HttpServer class to implement a web ui for configuration. Sometimes the browser stucks on the middle of the request. I tracked the problem to:
`HttpParser::parse(Socket s)` 
line 193:` line = s.readToDelim(lineTerminator)` -> `std::string Socket::readToDelim(std::string delim)`
It seems that the read from socket enters an infinite while loop waiting for the delimiter. I will investigate further but it occurs on random requests.

nkolban, 2017-12-27:
Would you be able to capture a trace of the browser traffic sent to the HTTPServer that causes the stall?  When using chrome, we can switch to the developer tools and see the trace of the data being sent.  Failing that we can always insert a TCPMon trace between the browser and the ESP32 or use WireShark to see the packets at the lowest level.

Stormhand, 2017-12-27:
Unfortunately im testing it on the phone and desktop pc w/o Wifi today. I tried on two different browsers.

nkolban, 2017-12-27:
I'm imaging an ESP32 hosting an HTTPServer with some HTML pages being served.  I'm now imagining your phone connecting to the ESP32 with the ESP32 being a WiFi access point.  I'm now imagining a browser on your phone navigating to a URL hosted by the ESP32 ... and at this point we get stuck.  Does "chrome" on the phone have a developer tools mechanism to examine the traffic?

Stormhand, 2017-12-27:
Thats what I meant but i dont use dev tools on the phone because it's not very comfortable on touch screen and small screen. I can investigate it tomorrow on the laptop.

Stormhand, 2017-12-28:
Well the part which causes the whole server to block is the User-Agent header in Chrome. I cant reproduce it in Mozilla so i replaced the headers. It still accepts an additional socket but works. The difference is in the line ending of the request headers of the browsers. Chrome adds 0D 0A, Mozilla  only 0A

Stormhand, 2017-12-28:
Sorry accidentally closed the issue :)

Stormhand, 2017-12-28:
Its very strange. Chrome is definitely making two different connections:

> No.     Time           Source                Destination           Protocol Length Info
>       1 0.000000000    192.168.100.7         192.168.100.80        TCP      74     59194 → 80 [SYN] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=1748960187 TSecr=0 WS=128
> 
> Frame 1: 74 bytes on wire (592 bits), 74 bytes captured (592 bits) on interface 0
> Ethernet II, Src: AsustekC_06:5e:c1 (30:5a:3a:06:5e:c1), Dst: Espressi_45:4f:ac (30:ae:a4:45:4f:ac)
> Internet Protocol Version 4, Src: 192.168.100.7, Dst: 192.168.100.80
> Transmission Control Protocol, Src Port: 59194, Dst Port: 80, Seq: 0, Len: 0
>     Source Port: 59194
>     Destination Port: 80
>     [Stream index: 0]
>     [TCP Segment Len: 0]
>     Sequence number: 0    (relative sequence number)
>     Acknowledgment number: 0
>     Header Length: 40 bytes
>     Flags: 0x002 (SYN)
>     Window size value: 29200
>     [Calculated window size: 29200]
>     Checksum: 0x49d7 [unverified]
>     [Checksum Status: Unverified]
>     Urgent pointer: 0
>     Options: (20 bytes), Maximum segment size, SACK permitted, Timestamps, No-Operation (NOP), Window scale
> 
> No.     Time           Source                Destination           Protocol Length Info
>       2 0.000163681    192.168.100.7         192.168.100.80        TCP      74     59196 → 80 [SYN] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=1748960187 TSecr=0 WS=128

The second connection from port  59196:
> Frame 2: 74 bytes on wire (592 bits), 74 bytes captured (592 bits) on interface 0
> Ethernet II, Src: AsustekC_06:5e:c1 (30:5a:3a:06:5e:c1), Dst: Espressi_45:4f:ac (30:ae:a4:45:4f:ac)
> Internet Protocol Version 4, Src: 192.168.100.7, Dst: 192.168.100.80
> Transmission Control Protocol, Src Port: 59196, Dst Port: 80, Seq: 0, Len: 0
>     Source Port: 59196
>     Destination Port: 80
>     [Stream index: 1]
>     [TCP Segment Len: 0]
>     Sequence number: 0    (relative sequence number)
>     Acknowledgment number: 0
>     Header Length: 40 bytes
>     Flags: 0x002 (SYN)
>     Window size value: 29200
>     [Calculated window size: 29200]
>     Checksum: 0x49d7 [unverified]
>     [Checksum Status: Unverified]
>     Urgent pointer: 0
>     Options: (20 bytes), Maximum segment size, SACK permitted, Timestamps, No-Operation (NOP), Window scale

The last tracks of 69196 are in 
Frame 6: 80 → 59196 [SYN, ACK]
Frame 7: 59196 → 80 [ACK]

Could anyone confirm that chrome opens two connections?

Stormhand, 2017-12-28:
The socket now properly timeouts:

> ------> new connection client 192.168.100.7:36584
> ------> new connection client 192.168.100.7:36586
> E (15611) Socket: receive: No more processes
> E (20611) Socket: receive: No more processes
> E (20751) HttpServerTask: Unable to open file /spiffs for reading

Please review for any mistakes. The methods are direct copy/paste from https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.cpp

squonk11, 2017-12-29:
I updated in my project to your latest commit. After this the Websocket detection seems to be broken (if I don't have another issue). Can you check that?

Stormhand, 2017-12-29:
It seems that the timeout kicks when the websocket is not active. I should change it to use bigger values for websocket. Or not set it at all for it?

Stormhand, 2017-12-29:
Created a pull request. The timeout is set to 0 if the request is a WebSocket

nkolban, 2017-12-29:
Merged.  My intent as the defacto moderator is to merge pull requests as soon as they become known to me.  Through Git, we can always back them out should we agree that there was a mistake.  The last thing I want to do is become a bottleneck for contributors.

squonk11, 2017-12-29:
No problem for me. I think it makes much sense to merge as soon as possible; in this way the progress is faster.
by the way: if you have time could you take a look at my recent question in issue: #286 ?
Kind regards

closed 2018-01-04 by Stormhand