nkolban/esp32-snippets #325: How to use MTU in esp32

chegewara, 2017-12-28:
By design i wanted to make it as much easy to use as its possible. This is code that needs to be added in application (server or client) to setup mtu:

`BLEDevice::setMTU(value);`

This piece of code needs to be added right after `BLEDevice::init()`. Thats all and that should works.

Value can be any unsigned integer between 23 and 517, this is bluetooth 4.2 specification limitation.

d3cline, 2017-12-29:
I have been debugging an ios related bug for 2 days which I am almost positive requires MTU to be set to 23, I wonder if these changes will fix it or effect it. Pretty ironic you were working on the fix.

chegewara, 2017-12-29:
@d3cline MTU 23 is default value and dont require anything to do with your code to use it.

d3cline, 2017-12-29:
Not on ios it isn't. And ios does not have a way to send MTU to renegotiate. The device must send the packet.

d3cline, 2017-12-29:
Basically characteristic value read on ios only shows the first 23 bytes. After extensive debugging and reading I determined it was the MTU. I did this by adjusting MTU on my android and watching the same behavior as well as the Debug log on the ESP itself. Unless MTU is set in the device to 23 the ESP side of the chunking fails. Since MTU can be set by the client and the device via a packet, it is the device which must send that packet, as they provide no way into the MTU via corebluetooth library.

d3cline, 2017-12-29:
P.S. I can't verify this yet, since apple has no MTU setting, and I was digging in this libraries code when I saw your ticket and knew it was related. I want to test this, but I am waiting for these changes to be merged into an official release as the apple devices are in portland, I am in colorado, and the CEO has to order some parts before he can re-flash the ESP32. But perhaps now explained it will make some sense.

chegewara, 2018-01-07:
@d3cline Do we have some progress here?

d3cline, 2018-01-08:
Yes I do! For the last few days I have worked on this, starting with a custom checkout of that part of the code which enables MTU, it did seem to send the packet. We got a reply from apple and it boiled down to the only way 'long reads' will work on apple devices is with BLE 4.2 commands. That even if you set MTU it won't matter. So I decided to manually chunk the data prior to send with an array and an iterator and this works for me. 

This was Apple's reply to our inquiry about this stuff ,

> 
> Unless your hardware is able to use BLE 4.2 Data Length Extension (only available on iPhone 7 and newer devices on iOS side), BLE transmissions are limited to 23 byte data packets over the air.
> DLE is different than MTU. Even if you increase the MTU, actual data packets are still limited (unless you are extending them with 4.2 features). All this needs to be done on the peripheral. There is no way for your app to control any of these low level settings.
> 
> When your data is larger than what fits in an over the air packet, you must split it into smaller chunks and send in separate packets. I cannot answer how this would be done on your specific hardware, and you may want to consult the support channels for it to find out how you would do this with the hardware and possibly the firmware you are using.
> 
> Sending multiple packets may or may not be trivial depending on how performance sensitive you are. If you are using “write with response”, you can send a single packet per connection interval, and you will use the next interval for the response, and so on.
> 
> If you are using “write without response”, you can send multiple packets within a single connection interval. How many you can send in a single interval will depend on your hardware and the connection interval you select. Most hardware can send 3. Some can send 4. Some will claim to be able to send 6, but that should be taken with a grain of salt.
> 
> If your hardware supports 4.2 DLE, and you can live with supporting iPhone 7 or later, using an extended data packet would be the best solution, as it will be future-proofing your solution.

I know we are chunking the data already but for whatever reason it has issues which manual chunking for me did not have. 

```
int snum = 0;
class MyCallbackHandler: public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic) {
    String SEND = String(snum)+":"+String(readings[snum]);
    SEND.toCharArray(vsendstr, 23);
    pCharacteristic->setValue(vsendstr);
    ++snum;
    if(snum == 12){
      snum = 0;
    }
  }
};
```

chegewara, 2018-01-08:
Thaks for help with this issue. I will ask espressif devs about it and if its supported in esp-idf will try to add support into library.

chegewara, 2018-01-08:
@d3cline Just to be clear, iPhone has to follow bleutooth specifications. That says it means iPhone shall support at least minimum MTU described by bluetooth core specification which is 23 bytes PDU. Im not saying iPhone can negotiate its size. I just found this document and i will study it:
https://developer.apple.com/hardwaredrivers/BluetoothDesignGuidelines.pdf

klynton, 2018-01-08:
@chegewara The iOS developer here! The iPhone does indeed support the specification and gets the 23 bytes. The problem is it only gets 23 bytes. The android BLE library will read 23 bytes at a time until all the bytes have been read but the CoreBluetooth only gets the first 23 bytes.

Does the ESP 32 support 4.2 Data Length Extension? Maybe we missed some configuration options on the firmware side that would enable us to do it easier.

chegewara, 2018-01-08:
@klynton Thank you for your interest. At the moment i think esp32 does not support Data Package Length Extension, at least i didnt find any info about it. We are waiting official info from developers. It is something new to me and now when i know what requirements we need to satisfy we will try to implement it if/when it will be possible.

d3cline, 2018-01-08:
I just want to specify a few things for clarity. Klynt and I work for the same startup and live in different states. I am the hardware/android dev, he is doing iOS. 

The bug is: I can use a chararray longer then 23, and BLECharacteristic.cpp will chunk it up and Android reads it fine. iOS only reads the first 23 bytes. 

To get around the bug I am now only sending across 23 bytes so the chunking does not happen in BLECharacteristic.cpp and it sends it once. 

I believed the iOS half of the bug has to do with the MTU packet, needing to be sent from iOS.

We still have not tested increasing MTU from the device to send to iOS to see if it can chunk things at the l2cap layer and not the characteristic. I did test if your code can send the packet and I can at least see it in the debug output in the arduino IDE. This is a very hard error to debug as I don't have physical access to the device nor do we have access to a BT sniffer. We are waiting to get a USB-TTY cable to Klynt so he can update the firmware of the device I shipped him and we can try a few things in real time. All this makes me wonder, is there a decent ESP32 BLE sniffer firmware for debugging, so I can see real packets in the air?

chegewara, 2018-01-08:
Can you provide logs when you send long data (mora than 23 bytes) to iOS? I would like to see how iPhone behaves.

klynton, 2018-01-11:
Here's the ticket I sent into Apple. There's some debugging info in it:

I'm having a problem with CoreBluetooth where it is only able to receive a small part of the data I'm transmitting. The device is a custom built spectrometer which has 2 characteristics from 2 different on-board sensors. The size of the values can be from 44 to 100 bytes. 

After a long time looking for Swift Bluetooth sample code that works I was able to get some data but only 22 bytes of it. Apple's Bluetooth sample code hasn't been updated since 2013 and there are no Swift samples for CoreBluetooth.

Using PacketLogger I was able to find the device being read but as you can see the length is (23):

http://i.heliost.at/Untitled_Live_2017-12-28_18-26-17.png

I'm aware of l2cap packets and I see that they are being used here but I'm, still, only able to see a small amount of the data.

Using the Apple Bluetooth Explorer I was able to find the device and values but they still are not the full values, they are truncated:

http://i.heliost.at/Low_Energy_Devices_2017-12-28_20-20-01.png

I've tried a few 3rd party Bluetooth explorers from Adafruit and Nordic Semiconductor. I was not able to find the full value from my device on either of them.

Using the PacketLogger app and I was able to capture the BLE stack reading the values from my device a few times:

http://i.heliost.at/Untitled_Live_2017-12-28_18-26-17.png
http://i.heliost.at/Untitled_Live_2017-12-28_18-03-12.png

During the capture I saw a few devices negotiate the MTU which I could not find any documentation about. (There were actually about 300 forum posts over the last 5 years asking how to do it with no resolution. There are hundreds of posts asking how to get data longer than 23 bytes via bluetooth on iOS.)

I created a virtual peripheral with the LightBlue iPhone app and manually set the value to a reading from the actual device. I was able to get the full value, all 88 characters of it, from the Bluetooth Explorer app:

http://i.heliost.at/Low_Energy_Devices_2017-12-28_18-37-04.png

I then followed this with the PacketLogger app where I watched all of the data get read from the virtual peripheral:

http://i.heliost.at/Untitled_Live_2017-12-28_19-57-58.png

So there isn't a problem with the size of the data, rather there is a problem with the way the device is being treated by CoreBluetooth or some code I'm missing that reads additional chunks.

Reading through the very sparse bluetooth "docs" on developer.apple.com I wasn't able to see anywhere to specify the length of chunks or even reassemble or read more data.

chegewara, 2018-01-11:
Hi @klynton Its nice amount of data, thank you. Ive found something interesting. Look at this screenshot:
http://i.heliost.at/Untitled_Live_2017-12-28_19-57-58.png

at 19:55:28.828 you have received packet, 45 bytes long and is received in one data transfer. Also data in this packet looks similar to data from this sreenshot:
http://i.heliost.at/Low_Energy_Devices_2017-12-28_18-37-04.png

Im guessing its exactly what you expecting to receive. If data should be longer then i dont see if more has been sent. At 19:55:28.850 we can see another request, maybe its request to send more but we cant see more.
There is one more packet i dont understand. At 19:55:26.499 we have device name request, but response is not human readable.
http://i.heliost.at/Untitled_Live_2017-12-28_19-57-58.png

klynton, 2018-01-11:
That data is the full amount of data, but it's from the virtual peripheral on my iPhone. The data from the ESP32 stops at 23 bytes.

chegewara, 2018-01-12:
Im thinking, maybe there is some event that is triggered during communication between esp32 and iPhone, but i dont have iPhone so im missing this part. Can you guys provide more serial logs from esp32 to let me see whats going on? Im interested in communication when ESP_GAP_BLE_ADV_START_COMPLETE_EVT is triggered and few seconds after iPhone connects to esp32.

bzeeman, 2019-04-25:
Is there a reason this issue is still open? It confuses me because I have used many packets exceeding the 23 byte "limit" with the ESP and iPhones pre-dating iPhone 7. Maybe I am not understanding what the issue is here but I consistently use packets of 187 and 513 to send firmware with no issues with data integrity.

closed 2019-04-25 by chegewara

jshep321, 2021-02-25:
I'm quite nterested in this also.  I'm using the nimble stack from @h2zero on an iphone 6S (iOS 14,4), but can't seem to get an MTU > 185.  Any resolution @bzeeman?
See my issue here: [https://github.com/h2zero/NimBLE-Arduino/issues/198](url)

chegewara, 2021-02-25:
Yes, trash iPhone and use normal smartphone.
It is iOS limitation, so how would you like to fix it?

jshep321, 2021-02-26:
> Yes, trash iPhone and use normal smartphone.
> 
> It is iOS limitation, so how would you like to fix it?

Well it actually works on iPhone 7 and later, but that's poorly documented by Apple.  MTU max of 185 on iPhone 6s.

dansoskin1, 2021-08-02:
for some reason it doesnt work for me,

i try to communicate with 2 esp32 devices, one as a server and other as a client.
from what i understand, i need to add the BLEDevice::setMTU(value); command right after the BLEDevice::init("")
only on client side.
but nothing seem to change, the packet still cuts at 20 bytes

(when connecting to server via nrf, i can change the mtu there and it sends more data to my phone, but nothing changes on the esp32 client)

what am i doing wrong? what am i missing?
im using the exmaple->client code, with the additional BLEDevice::setMTU(value); command

dansoskin1, 2021-08-05:
no iphone/ios, just 2 esp32s trying to communicate, one as a server and the other as a client

jshep321, 2021-08-06:
Ah ok.  I don't have that setup so can't test.  Let us know what you find!

chegewara, 2021-08-06:
https://esp32.com/viewtopic.php?f=19&t=22574