nkolban/esp32-snippets #327: bluetooth devices mac address find using esp32

mahdikan, 2017-12-30:
hello every one 
i want to search bluetooth macs by esp32 in arduino IDE but i couldn'e find any program to use ,
if you know what should i do please tell me .
thanks 
mahdi

nkolban, 2017-12-31:
Howdy,
Might this be a sample that you could use?

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleScan.cpp

mahdikan, 2018-01-03:
thanks to mr kolban
i taste this program but it dosent work.
i want to search mac address of devices around with enabled bluetooth 
like classic bluetooth

mahdikan, 2018-01-03:
i read in espressif web forum that the classic bluetooth is'n work in sdk yet , do you have any idea about that.
please chek this address:
https://www.esp32.com/viewtopic.php?f=13&t=1209&start=10

chegewara, 2018-01-03:
From my knowledge bluetooth classic is working on esp-idf but for sure its not working with this library. Its just not implemented yet. Thats why it is BLE library which means bluetooth LE.