nkolban/esp32-snippets #329: WebSocketHandler needs to know the calling WebSocket

squonk11, 2017-12-30:
I am now trying to do something useful with the WebSocketHandler. For this I implemented a MyWebSocketHandler. Inside the onMessageHandler I would like to send something back to the browser. But the WebSocketHandler class does not know the WebSocket. So, shouldn't there be a function like `WebSocketHandler::setWebSocket(WebSocket *pWebSocket);` 
in order to be able to use pWebSocket->send(...) inside the onMessage handler? Or do you have another solution for this?

nkolban, 2017-12-31:
Good question ... I had a look at the code and said to myself "That seems rather obvious, why did I not do that in the first place?" .... and then I thought some more and remembered my thinking.  Let me run this past you and see how it feels ...

The notion behind a callback handler is that you are responsible for implementing it.  For example, you would code:

```
class MyHandler: public WebSocketHandler {
public:
   void onMessage(WebSocketInputStreambuf *pWebSocketInputStreambuf) {
      // Your code here
   }
}
```

Does that make sense so far?

If so, then lets look at that again.  The class `MyHandler` that you just coded can have any additional properties and methods it wants ... for example:

```
class MyHandler: public WebSocketHandler {
private:
   WebSocket *m_pWebSocket;
public:
   MyHandler(WebSocket *pWebSocket) {
      m_pWebSocket = pWebSocket;
   }
   void onMessage(WebSocketInputStreambuf *pWebSocketInputStreambuf) {
      // Your code here
   }
}
```

which means that when you create your WebSocketHandler ...

you could:

```
WebSocket *pMyWebSocket = ...;
...
MyHandler *pMyHandler = new MyHandler(pMyWebSocket);
pMyWebSocket->setHandler(pMyHandler);
```

This means that you (the solution builder) can expand the WebSocketHandler in any way you wish ... such as saving/storing the WebSocket.

nkolban, 2017-12-31:
Alternatively ... and what I just change ... we now have an additional parameter on the `onMessage` method.  It now has the following signature:

```
void onMessage(WebSocketInputStreambuf *pWebSocketInputStreambuf, WebSocket *pWebSocket);
```

changed from

```
void onMessage(WebSocketInputStreambuf *pWebSocketInputStreambuf);
```

Please review and see if either of these are satisfactory stories and, if not, don't hesitate to let us know.

squonk11, 2017-12-31:
Initially I tried the solution you suggested in your first answer: implement a setWebSocket function in my handler class. But here I ran into some strange compilation errors - maybe due to some higer sophisticated C++ restrictions which I do not understand (like I do not understand why I cannot easily use a class member function as a callback function).
With your new onMessge handler function it now works as desired. 
Thank you for that!

nkolban, 2017-12-31:
That's great news.  Thanks for taking the time to respond.  When you are satisfied that this issue is resolved to your desire, please go ahead and close it.  We review open tickets and see if they need attention so if a ticket has been resolved, we like to close it out so we don't spend time reviewing resolved tickets.  We also don't like to presume that a ticket is closed until the original author is completely satisfied.

squonk11, 2018-01-02:
Issue is solved

closed 2018-01-02 by squonk11