nkolban/esp32-snippets #330: Suggested addition to BLEServer for requestMtu callback

rweller59, 2017-12-30:
I've been trying to get mtu increase working and found this helpful. Line numbers are approximate since my version is a couple of weeks old
BLEServer.cpp
(line 294)
		case ESP_GATTS_MTU_EVT: {
			if (m_pServerCallbacks != nullptr) { 
				m_pServerCallbacks->onMtuRequest(this);
			}
			break;
(line 419)
void BLEServerCallbacks::onMtuRequest(BLEServer* pServer) {
	ESP_LOGD("BLEServerCallbacks", ">> onMtuRequest(): Default");
	ESP_LOGD("BLEServerCallbacks", "Device: %s", BLEDevice::toString().c_str());
	ESP_LOGD("BLEServerCallbacks", "<< onMtuRequest()");
} // onMtuRequest

BLEServer.h
(line 118)
virtual void onMtuRequest(BLEServer* pServer);

chegewara, 2017-12-30:
Sure, why not. But may i ask what it can be user for?

rweller59, 2017-12-30:
So far I'm using it for troubleshooting since I'm having some issues getting the requestMTU to work. I'm assuming that there will be negotiation on mtu size depending on connected device. Reading large value length seems to be working without having to do additional work. Writing not so much - so I need to understand the mtu in order to size packets on server (esp) characteristic updates.

My understanding is that increasing mtu effects both send/receive length but haven't been able to make that work to date

chegewara, 2017-12-30:
Well, there may be bug in library code, so if you could share logs with use we could try to figure it out together.

chegewara, 2017-12-30:
@rweller59 I have a question, are you trying to connect with windows PC, disconnect and then connect with some other device like android?

rweller59, 2017-12-31:
Just updated espidf and cpp_utils a couple of hours ago and have been working with code that's a couple of weeks old if anything has changed around mtu. Outbound (from esp server) values were getting truncated previously. I will remove the 20 byte packet logic and see what's going on now. Even so, from what I understand, different devices have different max mtu values so seems like both client and server will need to know what mtu they are working.

rweller59, 2017-12-31:
Looks like esp is no longer truncating on outbound values - at least no indications. Not receiving the longer values on client side though so I'll have to try and figure out where things are breaking down.

rweller59, 2017-12-31:
Ok, looks like things are working with bigger mtu - inbound and outbound.

I'll close this but could you comment... if different client devices have different max mtu values, and I want to send/receive values with a length greater than the clients's max mtu, how would this work? Is there a way to get the mtu so that values can be chunked to not exceed mtu?

closed 2017-12-31 by rweller59

chegewara, 2017-12-31:
During connection peer divices are exchanging information about max MTU value they can work with. When this info is exchanged then both devices will set lower value as MTU used to communicate with each other. If characteristic needs to be read or write and value of that message is longer than MTU value -2 if i remember, then message is split into a parts and send in few part. This is called segmentation. Then all parts are connected after receiving (defragmentated). Only issue can be with notifications/indications, they cant be segmentated and are truncated to MTU-3 value.

chegewara, 2017-12-31:
@rweller59 There was bug related to MTU but i think its solved with the latest patch, i have to change one more thing and its waiting to make PR