nkolban/esp32-snippets #331: HttpServer::stop() needs to stop/end its running HttpServerTask

yaoyogi, 2017-12-31:
Currently stop() only does m_socket.close(), it doesn't stop the task that was created to actually run the server.

The public facing HttpServer class could hold onto the created task reference and then end it gracefully when stop() was called.  The other option could be to find the task by its name "HttpServerTask".  FreeRTOS has xTaskGetHandle(qry) which could be used to do the lookup.

yaoyogi, 2017-12-31:
A brief update.  Since Kolban Task is a wrapper on FreeRTOS task, the lookup gives us the FreeRTOS task but not "the" wrapper.  So I took the route where the HttpTask* is held onto by HttpServer.  On HttpServer::stop(), after the close(), I delay a short bit to let it complete, and then call the stop() on the wrapper task and then do a delete to free its memory.  Unfortunately Task::stop() hangs on its ::vTaskDelete(temp) call and just never returns.

nkolban, 2017-12-31:
Howdy my friend.  The way that the HttpServer works is that it creates a socket that is listened upon for incoming client requests.  This is performed in the background in its own task.  When  we wish to stop the HTTP Server, we call the stop() method.   This much I think you have fully realized.

The stop() method closes the socket that the HttpServer is listening upon.

Now consider this code which is the implementation of the HttpServer task:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp#L164

What it does is perform a blocking accept() sockets call.  This will normally wake up when a client forms a new connection.  However, if the socket on which the server is listening is closed (eg. through a stop() call) that will also cause the end of the accept() which should result in the listening task itself coming to an end.

yaoyogi, 2017-12-31:
Thanks for explanation.  I am a week or so behind on cpp_utils updates and see that some timeout was added to the socket accept() and used inside the HttpServerTask.  Having the timeout is useful functionality but will not solve the issue I am noting. 

The close() did not seem to unblock the accept().  I put a debug statements in both parts of the try/catch around the accept() and don't see either get called after the close() is called.  Note this is without the timeout code recently added, but that shouldn't change this behavior.  This is the initial behavior I opened this issue on.  Perhaps this is the real issue. 

You are observing that when you call close(), the accept() blocking, unblocks right after that call and the task ends?  I don't see this happening.

Some more about why the timeout approach is problematic for general web server processing:

As a webserver I can't say when a client is going to connect or how long between those requests.  

If a set a short timeout, say 5 seconds, and the accept() fails then the HttpServer task will end unless a client attempts to connect within 5 seconds.  Clearly not the scenario wanted for a long standing webserver.  

If I set a long timeout say 500 seconds, I run into the following set of problems when I call close().

1. The server is still bound to whatever port it started on, basically that port is not available for other communication.  
2. An incoming request will get handled by a server that I closed long time ago.  I believe another user had a similar issue and you added the recent timeout patch to address his condition.  Again, maybe ok for WebSocket'ish behavior but not for regular web http behavior.
3. Until a request comes along, case #2, or the accept timeout is reached, the task is still consuming quite a bit of memory as the HttpServer is created with quite a bit of stack.

I will reverify these observations after I pull your latest.

PS: The code is currently using a socket accept() vs. select() or such.  This single-threads the client requests vs. dispatching each request to a separate task or queue/task-pool style of processing. 

Thank you.

nkolban, 2017-12-31:
Howdy my friend.  The timeout code extension was an addition by another community member so haven't had a chance to study it in depth yet.  However, I ran the following code snippet to test HTTP server shutdown:

```
pHttpServer->start(9080);
ESP_LOGD(LOG_TAG, "Sleeping 5 seconds and then ending the server");
Task::delay(5000);
pHttpServer->stop();
ESP_LOGD(LOG_TAG, "Server stopped!");
```

The principle here is that we start the server, wait 5 seconds and then stop it.

Examining the serial logs, I find the following:

```
D (8051) task_httpserver: Sleeping 5 seconds and then ending the server
D (8052) Task: >> runTask: taskName=HttpServerTask
D (8062) Socket: >> setSSL: No
D (8065) Socket: >> listen: port: 9080, isDatagram: 0
D (8070) Socket: >> createSocket: isDatagram: 0
D (8075) Socket: << createSocket: sockFd: 8192
D (8079) Socket: >> bind: port=9080, address=0x0
D (8084) Socket: << bind
D (8087) Socket: << listen
D (8089) HttpServerTask: Listening on port 9080
D (8094) HttpServerTask: Waiting for new peer client
D (8099) Socket: >> accept: Accepting on 0.0.0.0 [9080]; sockFd: 8192, using SSL: 0
I (10888) wifi: pm start, type:0

D (13058) HttpServer: >> stop
D (13058) Socket: close: m_sock=8192, ssl: 0
D (13058) Socket: Calling lwip_close on 8192
E (13059) Socket: accept(): Software caused connection abort, m_sock=8192
D (13079) HttpServer: << stop
D (13081) task_httpserver: Server stopped!
E (13109) HttpServerTask: Caught an exception waiting for new client!
D (13110) Task: << runTask: taskName=HttpServerTask
```

Looking at the last few lines, we see the entry to the "stop()" function.  We see it close the listening HTTP server socket.  We see that cause the accept() to unblock with a return code that indicates an exception rather than a new incoming browser request and we end up ending the task that is the HttpServer.

Maybe if you can run with verbose trace enabled we can see what might be happening in your environment?

yaoyogi, 2017-12-31:
Thank you for quick reply.  I am pulling your latest and need to patch some other classes as I am week or so behind you.  In my current environment I don't see the "Caught exception waiting for new client!" and hence my current confusion/woes.  Hopefully we don't have some mismatch issues with versions of IDF causing some subtle differences.  I will retry shortly as you suggest.

nkolban, 2017-12-31:
No problems.  If you encounter any issues, PLEASE don't hesitate to get back to me.  I want to make sure you are as happy as I can make you :-)

yaoyogi, 2017-12-31:
What a wonderful philosophy!  Yes, and thank you again.

yaoyogi, 2018-01-01:
At this point it seems that even after the socket close() is called, it's possible the socket does processing for incoming requests.  This causes the behavior where the HttpServerTask does end but the port the socket was listening on is still bound and cannot be used again.  I have recreated the issue and will submit a standalone example later.   Note that if there are no incoming requests after the close(), the HttpServerTask does end and the socket completes cleanly allowing for HttpServer::start() to be called again.

nkolban, 2018-01-01:
We have just introduced some code changes for at least part of the puzzle.  When the HttpServer::stop() method is now called, it won't return until the server task has indeed stopped.  This should mean that a problem with:

```
server.start();
server.stop();
server.start();
```

shouldn't be an immediate issue because there shouldn't be a multi-tasking race condition following a `stop()` and the next `start()`.

Lets see if this makes progress.

nkolban, 2018-01-01:
Just a thought ... but it might be that the environment wide setting called "Enable SO_REUSERADDR" option might be what you need.

See "make menuconfig > Component config > LWIP > Enable SO_REUSEADDR option".

This is a system wide setting and not one that can set on by a component like cpp_utils.  However, someone using cpp_utils (like yourself) can switch it on at the "make menuconfig" level while building your project.

The snippet of help that comes with this option reads:

> Enabling this option allows binding to a port which remains in  TIME_WAIT.

yaoyogi, 2018-01-01:
Good idea, and yes, these settings can affect things.  
However I already had "Enable SO_REUSEADDR" and "SO_REUSEADDR copies ..." set.

Currently the HttpServerTask run() will do a m_socket.listen(...) each time.  While we get a new HttpServerTask each start() we don't get a new HttpServer instance and thus the m_socket member is shared across each start()/stop().  
On a subsequent start(), that same m_socket is being asked to listen on the same port again; this will cause the rebind error to occur and the listen socket is then dead for requests.  Thus the HttpServerTask::run() preamble should not redo the socket listen.

There is a general outline discussion for a similar problem at: https://os.mbed.com/forum/helloworld/topic/2572/  

Another potential issue with is that start(port, bool) means it's possible for me to try to rebind the same port of SSL and non-SSL with the same HttpServer instance.  Probably don't want to allow that.

yaoyogi, 2018-01-02:
Well this rebind problem looks like a general problem with sockets.  While there may be some lifecycle issues with the HttpServer/Task api's we have to get socket.close() to not just close the socket connection but close the socket as well.  This looks like a TIME_WAIT problem is possible this point and a reuse-addr.  I am experimenting with various socket settings on the listen socket to see how quickly we get to a "complete close" state and are able to relisten/rebind to the server port again.  

<Update> I was able to specifically set the listen socket to SO_REUSEADDR.  This cleared up the reuse error.  However I have a bunch of other tweaks I was adding that I need to back out/re-evaluate as to the fix.  I also needed to add some defensive code in the HttpServerTask::run() which I need to reverify again...  I was also trying to set SO_LINGER but ran into some problems here.  I don't know if this will be a problem if we have a large response still on the network stack and try to do the close(), ideally the resuse_addr will let us get past this scenario...

yaoyogi, 2018-01-02:
Ok, I have a fix and test/example for the "rebind port issue" and "connects coming in during close problems".  

Unfortunately the last update for start()/stop() serial behavior is imcompatible...  I will need to check out why but I think it's the HttpServerTask run() has defensive code for the connect coming in during close problems and probably short-circuits the semaphore that was added.

yaoyogi, 2018-01-02:
Ok, I have a fix for the "rebind port issue" and "connects coming in during close problems".  

I would like to submit these attachments for your review.  I commented out the recent start()/stop() semaphore code since it was causing some problems with the fixes I added.  Hopefully its not a big deal, but I would really like to get the resuse addr and connects during/after close issues working first, and then revisit the start/stop synchronization perhaps on a separate issue.

Socket.h::listen - line 77, listen(..) gets another argument for reuseAddr.  Since its defaulted to false existing clients should not have to change their code or deal with any semantic changes.

Socket.cpp::listen- lines 261-265.  This does the resuse addr if the new arg is true.  Since the new are defaults to false, all existing behavior will not change.  I needed to add the extra arg since inside listen() is where the createSocket(isDatagram) is done and immediately following is the bind(...), thus no other place to get ahead of these ...

HttpServerTask::run- 
    	lines 163:169 and 174:179.  These are defensive checks based on testing where the close() cycle and incoming connections was causing problems.
	line 159.  This uses the reuse addr option to create the listener socket.

[HttpServer.cpp.txt](https://github.com/nkolban/esp32-snippets/files/1596979/HttpServer.cpp.txt)
[Socket.h.txt](https://github.com/nkolban/esp32-snippets/files/1596980/Socket.h.txt)
[SockServ.cpp.txt](https://github.com/nkolban/esp32-snippets/files/1596981/SockServ.cpp.txt)

Thank you for your investigations and consideration of these changes.

yaoyogi, 2018-01-09:
The current code fixes the original issue I opened so I will close this.  The other issues I ran into about reuse addr and close(), will be opened as a new issue.

closed 2018-01-09 by yaoyogi