nkolban/esp32-snippets #332: JSON setBoolean/getBoolean

HanVertegaal, 2017-12-31:
I am probably just not seeing the obvious, but the following code has me wondering:

`JsonObject json = JSON::createObject();
json.setBoolean("Test", true);

ESP_LOGI(LOG_TAG, "Test=%s", (json.getBoolean("Test") ? "true" : "false"));

JSON::deleteObject(json);`

nkolban, 2017-12-31:
Building a unit test now for debugging to see what might be happening.

chegewara, 2017-12-31:
`ESP_LOGI(LOG_TAG, "Test=%s", (json.getBoolean("Test") ? "true" : "false"));`
Im not sure but you are testing here value or existence of object with "Test" key?
Maybe try with this:
`ESP_LOGI(LOG_TAG, "Test=%s", (json.getBoolean("Test") == true? "true" : "false"));`

nkolban, 2017-12-31:
@HanVertegaal  ... Thank you sir.  Excellent report.  The problem was found and turned out to be a mistake on my part in understanding / studying on the underlying cJSON library.  I wish I could say that there was more to it than that ... but unfortunately, the blame lies squarely at my feet and it was purely sloppiness on my part.  The code has been corrected and the test you supplied which originally failed now works.  When you get a moment, please test with the latest cpp_utils and let us know if this has resolved the issue.  I apologize for wasting your time having to hunt this down.

HanVertegaal, 2017-12-31:
After pulling the latest changes to cpp_utils, I got I2C compile errors that were described in another issue. This prompted me to pull the latest changes on the v3.0 branch of the ESP-IDF, but that didn't help. Switching to the head of the ESP-IDF master branch resulted in a file not found error (esp_gap_ble_api.h).

Could you please tell me what version of the ESP-IDF you used to build cpp_utils? 
(It may be helpful to keep that information updated in the readme somewhere.)

I'm actually somewhat embarrassed that I did not take a closer look at the JSON class implementation myself, and propose a fix. I guess I was too convinced that it must be a simple oversight on my part.
Let me thank you for all your fine work on this library, which has surely saved me countless hours of coding.

nkolban, 2017-12-31:
This project (cpp_utils) really only started gaining momentum shortly after 2.0 of the ESP-IDF and there hasn't been a formal ESP-IDF release since then (we are on the cusp of 3.0).  When 3.0 ships, I plan to keep two branches of cpp_utils.  One which builds against ESP-IDF master and one which builds with ESP-IDF 3.0 releases.  However ... as of **this** very date and time .... I am testing cpp_utils exclusively against ESP-IDF master.

https://github.com/espressif/esp-idf

... within everything implied by that ... including the correct toolchain for that release.

Try and build with ESP-IDF master (as of today) and if something fails to build cleanly, post as much detail as you can about compilation errors found.

HanVertegaal, 2018-01-01:
Apparently I needed to have Bluetooth enabled for everything to compile smoothly.
See [https://github.com/nkolban/esp32-snippets/issues/335](url).

The JSON::getBoolean and JSON::setBoolean work fine now.

closed 2018-01-01 by HanVertegaal