nkolban/esp32-snippets #333: Socket::bind() and Socket::close() should have return code.

yaoyogi, 2018-01-01:
Currently there is no easy way to know if these calls fail, other than to:
1. see downstream code problems
2. look at the logs under debug
3. know to look at errno.  While errno is possible, it requires the person to look at the socket impl and look at what lwip calls are made and then go from there.  Since Socket wraps the lwip constructs, it would be nice to have those errors propagated and/or mapped to the return code. 

Maybe bind() could get away with just returning what lwip_bind_r(..) returns which is essentially lwip_bind()  Normal Linux bind(..) has a pretty rich error semantic so it's kind of tough for this call to be just void return.  For example, I can call bind(...) on the same port repeatedly and easily miss the error that the address is already in use (hence why I am opening this issue :)

close() abstracts the close over ssl and non-ssl, so maybe a simple ESP_OK and ESP_FAIL would suffice.

If they return esp_err_t, then ESP_ERROR_CHECK(..) programming model would work.  Even if bind() is just 0 or -1 you can at least programatically do something and if necessary go to errno for more details.

Since the current API's are void, it shouldn't be too much of a problem to have them return a value.

yaoyogi, 2018-01-01:
Please add Socket::listen(..) as to issue #333 as well.

nkolban, 2018-01-04:
Thank you sir.  All these make sense.  We have now added return codes to:

* Socket::bind()
* Socket::close()
* Socket::listen()

yaoyogi, 2018-01-04:
Thank you for quick response and update!  

These will help with my error handling code.

closed 2018-01-04 by yaoyogi