nkolban/esp32-snippets #334: cpp_utils Socket::accept() returns Socket from automatic storage

yaoyogi, 2018-01-01:
created: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Socket.cpp#L74

returned: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Socket.cpp#L85

nkolban, 2018-01-04:
I'm not sure what you are showing me here?

We do indeed create an instance of a Socket object in automatic storage (i.e. on the local scope stack), we then populate its socket file descriptor and we return a new instance of it.   The alternative would be to take as input a reference to an existing Socket object and populate it or create a new Socket object on heap storage and return a reference to it.  Of these three options, I chose to pass "copies" of the object because it is small enough.   Is there a bigger story you are seeing?

yaoyogi, 2018-01-04:
Apologies for not adding more details and perhaps my initial look at the code was not detailed enough.  
Yes the size is not too big; a sizeof shows: 1416 bytes.  However, the accept() is not just state setup.  In the case of SSL it actually does a receive() operation.  The bulk of the non-primitive members are also all SSL related, so after an operation I was unsure if returning a copy was what was intended.  I will close this as it is a confirmed design decision.  Thank you.

closed 2018-01-04 by yaoyogi