nkolban/esp32-snippets #335: Compile error with Bluetooth disabled

HanVertegaal, 2018-01-01:
When Bluetooth is disabled through make menuconfig, compilation results in the following error:

`In file included from /mnt/c/Users/Han/Documents/Projects/esp32base/components/cpp_utils/./BLESecurity.cpp:8:0:
/mnt/c/Users/Han/Documents/Projects/esp32base/components/cpp_utils/BLESecurity.h:10:29: fatal error: esp_gap_ble_api.h: No such file or directory
 #include <esp_gap_ble_api.h>`

It seems BLESecurity.cpp lacks a guard on CONFIG_BT_ENABLED.

nkolban, 2018-01-01:
Thank you sir.  We have made the fixes you identified.  Many thanks for reporting and especially many thanks for taking the time to investigate.   We apologize for the inconvenience that this caused.  I'd love to say we'll do better in the future ... but I'm quietly confident that we'll make these mistakes over and over and over :-)   Please review and close at your leisure.

HanVertegaal, 2018-01-01:
Compiling with Bluetooth disabled no longer generates error messages.

closed 2018-01-01 by HanVertegaal