nkolban/esp32-snippets #336: BLE example size

shaielc, 2018-01-01:
hi, just tried compiling the BLE_Server example on the arduino IDE,  and I get that it will use 69% of memory.

is that normal or is it a problem on my end?

nkolban, 2018-01-01:
Howdy,
I've never noticed such a message.  By any chance would you be able to post a screen shot or a text paste of the message so that I may see it in context?

shaielc, 2018-01-01:
Sketch uses 905182 bytes (69%) of program storage space. Maximum is 1310720 bytes.
Global variables use 62092 bytes (21%) of dynamic memory, leaving 232820 bytes for local variables. Maximum is 294912 bytes.

chegewara, 2018-01-01:
Yes, its normal. Bluedroid stack is large and BLE library is taking some space so almost clean app with bluetooth is taking about 900kB

shaielc, 2018-01-01:
thanks! is there anyway to decrease that if I intend on only using some features?

chegewara, 2018-01-01:
Now i dont see any way, but maybe you can say what you want to use and what we need to turn off?

shaielc, 2018-01-01:
I am only using it as a way to set a device remotely, basically can't use the WiFi as an AP to do that since I need the other device to have connection to an online source.

chegewara, 2018-01-01:
I think that bigger issue is how partition table is prepared under arduino. We have 3 partitions 1.3MB each and we cant do anything with that. Maybe we can change partition.bin with different but never tried it and this is question that should be issued on espressif/arduino-esp32 board.

nkolban, 2018-01-01:
To expand a little on the last post ... it may be you don't know about partitions?   The ESP32 has 4MBytes of flash space.  All of this is available to your application.  A configuration called "partitions" then describes how this 4MBytes is divided up for application use.  The default is that only 1MByte is allocated for program storage.  Why so small?   The designers of ESP32 imagine that a production application may, from time to time, need replaced "in the field".  This means that a new copy of the application can be downloaded over WiFi and replace the existing program.  Now imagine that you have a new version of your program that you wish to distribute in production.  You have to download that program, save it to flash and then when done, restart the ESP32.  However, when you are downloading and reflashing the ESP32, how should that be done?  You can "replace" your existing application but imagine that you have downloaded 50% of the program and replaced 50% of your original program and then the network failed.  You would now have "bricked" your ESP32 (from a user's perspective).  The solution is to partition the flash space.  For example, have TWO 1MByte partitions for programs ... only ONE partition is active at one time.  When you wish to replace your app, you download a new copy into the UN-USED partition and, only when it is complete, do you flag the new partition as the one to run from.    If you don't need this capability, you can change the partition size to be about 3.8MBytes and have only 1 partition ... and now your program space available for code will be nearly 4x what you have today.

shaielc, 2018-01-01:
so even if I don't use OTA updates the default is that it's partitioned?

chegewara, 2018-01-01:
Yes, thats how arduino-esp32 is configured (just guessing from observations). You should have ask about partitions here https://github.com/espressif/arduino-esp32/issues or on forum. You can also get 'normal' partitions.bin from someone and flash it with flash downloader tool

shaielc, 2018-01-01:
thanks!
will do.

pbecchi, 2018-02-05:
Coming back on your discussion, I would like to expose my case where the flash partition change solution is not practical.
I am trying to add Ble capabilities ( a Ble Uart client) to an existing code that use several libraries and that is already about 600 on flash size. I use Ota, Spiffs EEprom .... So I cannot change partition sizes!

The Arduino client example take about 950 on and my code with the Blue client snippet give this:

Program too big. The size is 1349762 bytes (of a 1310720 byte maximum). Ensure debugging is OFF and/or see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Build failed for project 'ESP32_OpenSprinkler'

My comments is that program partition size is already very large and that we should do something to reduce Ble library footprint!
Any guidance to change configuration files to set only the flags ( and the code ) necessary for my application?
For example : I don't need any server capability, since my app can only be a Ble client, any way to take that part of the libraries/stack out of compilation?

lmihalkovic, 2018-05-29:
No matter what the story is with the partitioning... the BLE library is just fat .. browsing through the code I think it was not devised with embedded in mind...

lmihalkovic, 2018-05-29:
I am thankful for the existence of this library.. immensely. It has an easy abstraction to work with.. so far does what i need (i ditched my own tinylib) .. and i still wish i was able to trim the lib rather than have to reshape the partitions as i did... everything can always be made better ... this library too .. 

thank you for the work done .. all who have contributed.

jjoe64, 2018-06-06:
@lmihalkovic can you public your own tinylib? I have the same problem with the size of this BLE lib and I think many others would also love to use a smaller version.

chegewara, 2018-06-06:
What i have found to day is that this value is hardcoded and is not always correct reported in arduino:
`Maximum is 1310720 bytes.`
With `no_ota` selected there is 2MB size app partition.
With 'default' we have 2 partitions 1310720 each.
With `minimal` is just one 1310720 partition.

jjoe64, 2018-06-08:
Hi all, I faced the same issue and did not found a easy solution, so I finally forked the library and create a "light version" of it.
Only the BLE server will work.  This is a fork of the official esp32 ble library and is stripped to decrease the size. It is only possible to use the ble server features.

I saved about ~30KB program size with this version.

[https://github.com/jjoe64/esp32_ble_light](https://github.com/jjoe64/esp32_ble_light)

chegewara, 2018-06-26:
There is upgraded configuration for arduino-esp32. You have new partition scheme: minimal SPIFFS.

ccfoo242, 2018-06-26:
@chegewara Is this something I can change so i can use this BLE library? My program isn't too large but adding BLE to it has pushed it past the 1.3MB limit.

ccfoo242, 2018-06-26:
@chegewara ok thanks. can you point me to some directions on how to change that? I'm using either Arduino 1.85 or whatever the latest platform.io is. Is this a build switch?

chegewara, 2018-06-26:
Go to 'C:\Users\USER_NAME\Documents\Arduino\hardware\espressif\esp32' and issue 'git pull'.
Then you should have new option in 'Tools->Partition scheme'.

ccfoo242, 2018-06-26:
Thanks, I will try that when I get home.

1technophile, 2018-06-26:
The partition scheme is available when you select "ESP32 Dev module"

ccfoo242, 2018-06-26:
In platioformio.ini I just had to add this line:
board_build.partitions = no_ota.csv

Thanks!

closed 2018-11-13 by chegewara

borch84, 2019-06-01:
Hi Everyone, I was able to strip my code down to 1002KB, I am using Bluedroid library + AWS IoT library for my purposes. I removed several options on bluetooth:
I removed debugging and also I am not needing any bluetooth  library because my ESP acts as GATT Server.
I hope this helps.
[image]