nkolban/esp32-snippets #337: advertised data

shaielc, 2018-01-01:
If i would like to change the advertised data how will I do that? haven't seen a way to change dynamically the advertised data

nkolban, 2018-01-01:
Howdy,
We just updated this document:

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

Which is a programmer's guide for the library.  Have a look at the section which references the `BLEAdvertisementData` and see if that assists.  If not, come back and we'll see what can be added to the docs to make it more usable.

shaielc, 2018-01-01:
thanks! didn't see it in the source code.

nkolban, 2018-01-01:
I'm always interested to hear feedback on consumption.  Don't hesitate to comment if you see something that could be improved or better described.

shaielc, 2018-01-01:
the function isn't anywhere to be found in the source code in
https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/src
wrong place?

shaielc, 2018-01-01:
seems like there's no implementation for adding Service Data, type 0x16.

nkolban, 2018-01-01:
checking now

nkolban, 2018-01-01:
There are so many advertising data types, we took the stance that we wouldn't just "add them all" until and unless someone asked for them.  We have now added advertising types 0x16, 0x20 and 0x21 all of which can be set using the method:

```
BLEAdvertisementData#setServiceData(BLEUUID, std::string)
```

We have a clean compile and *believe* we have done it right ... if you get a chance to test, please let us known.

shaielc, 2018-01-01:
testing now

shaielc, 2018-01-01:
`#include<BLEDevice.h>
#include<WiFi.h>
#define UUID_SERVICE "78489c82-5e4c-4c0b-b3ca-54390af7bf8c"
#define UUID_CHAR "12ff3aae-2399-4db7-a4c4-749d69127b7e"
BLEServer* setting;
BLEService* service;
BLECharacteristic * Char;
#define EDDYSTONE_PRE_HTTP_WWW 0
#define EDDYSTONE_PRE_HTTPS_WWW 1
#define EDDYSTONE_PRE_HTTP 2
#define EDDYSTONE_PRE_HTTPS 3

String eddystoneURL(byte prot,String address)
{//need to add POSTFIX dictionary
  String encoded ="";
  encoded += (char)0x10;
  encoded += (char)0x20;//should be TX power
  encoded += (char)prot;
  encoded+=address;
  Serial.print(encoded);
  return encoded;
}
void setEddystone(String address,BLEService * setting)
{
 BLEAdvertising *ad = setting->getAdvertising();//advertiser
 BLEAdvertisementData eddystone;//advData
 eddystone.setFlags(ESP_BLE_ADV_FLAG_GEN_DISC|ESP_BLE_ADV_FLAG_BREDR_NOT_SPT);//FLAGS
 eddystone.setCompleteServices(uint16_t(0xFEAA));//Google Eddystone UUID
 address = eddystoneURL(EDDYSTONE_PRE_HTTPS,address);//https://address coded to Eddystone service data
 eddystone.setServiceData((BLEUUID)((uint16_t)0xFEAA),address.c_str());//Eddystone UUID + addres for Eddystone
 ad->setAdvertisementData(eddystone);//set
 ad->start();//GO!
}
void setup() {
 Serial.begin(115200);  // put your setup code here, to run once:
 BLEDevice::init("Eddystone_example");
 setting = BLEDevice::createServer();
 service = setting->createService(UUID_SERVICE);
 service->start();
 String address =String("goo.gl/eGfeuj");
 setEddystone(address);

 Serial.print("fine");
}

void loop() {
  // put your main code here, to run repeatedly:

}`

nkolban, 2018-01-02:
Nice!!!

shaielc, 2018-01-02:
BTW : maybe it will be advantageous to implement a general advData function such as:

`BLEAdvertisementData:addData(uint8_t type,char* data,uint8_t length)`

for general usage.

closed 2018-11-13 by chegewara