nkolban/esp32-snippets #339: How to enumerate characteristics UUIDs

BigJBehr, 2018-01-02:
I am attempting to use the BLE Client on an ESP32 to connect with a bluetooth device that has a joystick and some buttons. I was able to discover the service UUID and change the service UUID in the example BLE Client. Now I can connect to the remote device. However, I do not know what the characteristic UUIDs are. Would you be so kind as to please provide an example of how to enumerate the characteristic UUIDs once a device is connected.

chegewara, 2018-01-02:
One quick question, is it standard bluetooth ble joystick? Which is recognised by windows as a joystick?

nkolban, 2018-01-02:
You bring up an interesting question ... one that I personally have never considered before and I'm not sure what the right answer should be ... so let's chat about it together.

Imagine we have an "alien" BLE Server.  This is a server that we know nothing about.  It is not an ESP32.  We now have an ESP32 client and wish to "work" with that BLE server.  I have assumed up till now that you would use a tool (eg. nRFConnect or bluetoothctl) to examine the BLE server and manually learn "what it is and what it does".  Typically, you are given a "specification" sheet of such a device and are told "This device exposes a service with UUID xyz and this service has characteristics with UUIDS abc, def and ghi".   You would then write an app that uses these values and you are good to go.

What I think I am hearing from you here is a desire to "dynamically" interrogate a BLE server and, at run-time, determine the IDs contained within it.  While this is technically possible (and in fact is already performed within the internals of BLE) I don't think we considered it as a capability to be externally exposed to BLE programmers.   We can most certainly change that.

However, before I go off into the darkness to implement some code, I want to understand in greater depth what it is you are looking for and how it would be used.  Would you be able to describe a "user story" or patterns where such a feature would be useful and how it would be used?

nkolban, 2018-01-02:
A colleague pointed out to me that we actually **already** have the ability to determine the characteristics.

When we connect to a remote BLE Server, we use a class called BLEClient.  Once we have a BLEClient object, we can call its `getServices()` method.  This returns a C++ map object that contains all the services exposed by the remote BLE Server.  This includes a BLERemoteService reference for each service.

From a BLERemoteService object, we can call its `getCharacteristics()` method which returns a map of all the characteristics associated with that service (remember .. there is a tree structure here).  A BLE Server can have 1 or more services and each service can have one or more characteristics.

The C++ map object might possibly be new to you.  Investing time studying the docs on C++ maps generically will be the way to go and then you can apply those skills in the BLE environment.

try a google using "c++ map iterator" for examples.

BigJBehr, 2018-02-09:
Thanks for looking into this for me. I will look into C++ maps. I have worked with C/C++ for decades and still do not know everything about C++. Seems like I learn what I need to get by and no more.

I found that an app on my Android device gives me all of the service UUIDs and characteristic UUIDs for each service. Using this information I was able to twist your BLE Client for the ESP32 to work with the hand-held mouse/pointer device I wanted to interface with.
The main service UUID is 0x1812 (HID device). This service has 10 characteristic UUIDs associated with it. Of those 10, six are reports (UUID 0x2A4D). I have handles for each of the reports.

New questions;
1 - How can I get the BLE Client to receive notifications from multiple reports under one service ?
2 - How can I get the BLE Client to use multiple services at the same time ?

Multiple notification is because different reports are reporting different sets of buttons an the device. No single report has all of the button data.
Multiple services because the HID Device service has the button reports and the Battery Service (UUID 0x180F ) has the battery voltage.

One last question, Is there any Arduino libraries for the ESP32 that allow working with classic Bluetooth ?
I have a Bluetooth gamepad that I would love to use, but it is Bluetooth 2.0 EDR, not BLE.
Thanks in advance.

chegewara, 2018-02-09:
Hi @BigJBehr 
I see you are trying to build esp32 hid host and i dont want to scare you at all, but when ive been building hid peripheral library ive found how hard it is. Its a lot esier if you want to create hid host for one particular peripheral because you know its data structure. Now back to your questions.
For each characteristic you can have function that will handle notification. So if you have 6 report characteristics you can have 6 functions to handle its notifications. It does not matter what service its belongs to because notifications are bonded to characteristic. If you have more questions i will be happy to answer here or on irc.