nkolban/esp32-snippets #345: How to read Eddystone TLM?

shahabmusic, 2018-01-05:
Is it possible to extract temperature values from an Eddystone TLM data?

I have a beacon that advertises the temperature and battery level. I can see these data using Android eddystone app and would like to see these data in esp32 in Arduino IDE too.

In one of the snippets I saw that you wait until you receive an Adv data type 0x16 and then you look for 0x1809 uuid and then you decode the temperature.

I tried to do the same thing in BLEAdvertisedDevice class. It find adv data type 0x16, however the esp32 does not find 0x1809 uuid. 

This is my serial output.
```
`[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT

[D][BLEUtils.cpp:1241] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: cc:78:ab:8d:7d:76, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -68, ble_adv: ??, flag: 4 ([BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 25, scan_rsp_len: 0

[D][BLEAdvertisedDevice.cpp:487] setRSSI(): - setRSSI(): rssi: -68

[D][BLEAdvertisedDevice.cpp:298] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 04

[D][BLEAdvertisedDevice.cpp:298] parseAdvertisement(): Type: 0x03 (ESP_BLE_AD_TYPE_16SRV_CMPL), length: 2, data: aafe

[D][BLEAdvertisedDevice.cpp:516] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000feaa-0000-1000-8000-00805f9b34fb

[D][BLEAdvertisedDevice.cpp:298] parseAdvertisement(): Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 16, data: aafe20000b3714000080a09101443b20
```
`

Am I in the right path to get temperature readings? could anyone help please?

chegewara, 2018-01-05:
This is your UUID `0000feaa-0000-1000-8000-00805f9b34fb`. Also check this line `[D][BLEAdvertisedDevice.cpp:298] parseAdvertisement(): Type: 0x03 (ESP_BLE_AD_TYPE_16SRV_CMPL), length: 2, data: aafe`. This is your data packet `aafe20000b3714000080a09101443b20` but you have to remember that bluetooth is using little endian.

chegewara, 2018-01-05:
But i think this data might be broken. This is nice info to read, first link i found with google:
https://github.com/google/eddystone/blob/master/protocol-specification.md

chegewara, 2018-01-05:
@shahabmusic You can check what raw data are used to advertise, i like to use nRF connect, and read from it what service UUID is used in advertising packet. If you have issue to read it you can paste here raw advertising packet or screenshot and we will try to help. Screenshot prefered like this one
https://ibin.co/3lNX93QvWwn3.png

shahabmusic, 2018-01-06:
here is screen shot of nRF

[https://imagebin.ca/v/3nEK1J5YY3FS](url)

chegewara, 2018-01-06:
Ok, this is what i see:
- its unencrypted eddystone
- here you can find specs and try to decode it https://github.com/google/eddystone/blob/master/eddystone-tlm/tlm-plain.md
- if you still have issue you may come back and teel us, we will try to help

shahabmusic, 2018-01-07:
Thank you so much. Finally I could solve the puzzle after 3 weeks. 
So in my raw data:
`Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 16, data: aafe20000b651b00008151d801667640
`
Temperature is 0x1b00. When I use a online hex to integer converter, I can convert b1 to 27 and 00 to 0 so temperature is 27.00 degrees. Now how can I do this conversion  in C?

chegewara, 2018-01-07:
Again, google is your friend. First link:
https://stackoverflow.com/questions/1070497/c-convert-hex-string-to-signed-integer

I like this solution:

>  Andy Buchanan, as far as sticking to C++ goes, I liked yours, but I have a few mods:
`template <typename ElemT> struct HexTo {     ElemT value;     operator ElemT() const {return value;}     friend std::istream& operator>>(std::istream& in, HexTo& out) {         in >> std::hex >> out.value;         return in;     } };Used likeuint32_t value = boost::lexical_cast<HexTo<uint32_t> >("0x2a");
`

nkolban, 2018-01-07:
Once you have the 14 bytes of TLM frame data, they can be decoded by mapping that data to the following structure:

```
struct tlm {
   uint8_t frameType;
   uint8_t version;
   uint16_t voltage;
   uint16_t temp;
   uint32_t pduCount;
   uint32_t timeCount;
}  __attribute__((packed));
```

see .... https://github.com/google/eddystone/blob/master/eddystone-tlm/tlm-plain.md#unencrypted-tlm-frame-specification

1technophile, 2018-01-07:
Hello,

Seems the kind of work I have done with mi flora : 
https://1technophile.blogspot.fr/2017/11/mi-flora-integration-to-openmqttgateway.html

In this file : 
https://github.com/1technophile/OpenMQTTGateway/blob/master/ZgatewayBT.ino

 you have methods to : 
* extract service data
* extract the interesting part of the service data
* revert if needed
* convert it to integer
* publish it to mqtt

In particular take a look to process_miflora_data.
Maybe not the most elegant way but works.

masdel123, 2020-06-23:
> Thank you so much. Finally I could solve the puzzle after 3 weeks.
> So in my raw data:
> `Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 16, data: aafe20000b651b00008151d801667640 `
> Temperature is [...]

How could you solve this problem?Can you explain this? Because I have same issue about tlm-parsing to get battery value