nkolban/esp32-snippets #346: pairing filure when own_addr_type is BLE_ADDR_TYPE_PUBLIC in the BLE advertise parameters.

JoonDong2, 2018-01-05:
I'm testing gatt_security_server example in the ESP-IDF.

original is set own_addr_type to BLE_ADDR_TYPE_RANDOM when configuring the BLE advertise parameters. At this time, ESP32 advertise adv packet including random generated address instead of MAC address. Because of this, android phone cannot recognize bonded device. In other word, android phone recognizes it as a new device every time ESP32 is rebooted.

So I configured own_addr_type to BLE_ADDR_TYPE_PUBLIC, then ESP32 advertised adv packet including MAC address always. connection is done, but pairing procedure is failed when I did this. The message is below. I did not revise other configures.

ESP_GAP_BEL_SEC_REQ_EVT
ESP_GAP_BLE_AUTH_COMPL_EVT
remote BD_ADDR : 222248398ac0
address type = 0
pair status = fail
Bonded devices number : 0
Bonded devices list : 0
btm_sec_clr_temp_auth_service() - no dev CB
Device not found
ESP_GATTS_DISCONNECT_EVT

Is this a bug? I want them to pair naturally after ESP32 rebooted.

chegewara, 2018-01-05:
Thanks for reporting, will investigate. If you can share code you are using it would be faster to investigate it.

chegewara, 2018-01-06:
Ok, i see the problem. When you changed to use public address (mac address) you also need to change this line to not generate random address by commenting it out or set false:
```
//generate a resolvable random address
            esp_ble_gap_config_local_privacy(true);
```

JoonDong2, 2018-01-06:
Thank you. It works on android nougat(7.1.1), but doesn't work on kitkat(4.4.2) when I configured to that.
On kitkat, pairing procedure is ok, but connection is destroyed immediately from second connection.
I'm not sure about other versions.

chegewara, 2018-01-06:
Yes, im aware of that becuse im having the same issue with my android 5.0 device. Its reported to espressif. Thank you for your reports.

JoonDong2, 2018-01-06:
On marshmallow(CM13.0), pairing procedure and re-connection is done, but connection state remains at "bonding..." in the nRF Connect app. Bonding information did not appear in the bluetooth setting. ESP32 save the pairing information.

chegewara, 2018-03-30:
Should have been fixed now.