nkolban/esp32-snippets #347: Compile Error on Linux Mint 18.3

kd8bxp, 2018-01-06:
I am trying to follow this project:
https://www.hackster.io/mitchwongho/read-phone-notifications-using-esp-eb0ad4?utm_source=Hackster.io+newsletter&utm_campaign=69cb508d3b-EMAIL_CAMPAIGN_2017_07_26&utm_medium=email&utm_term=0_6ff81e3e5b-69cb508d3b-135069861&mc_cid=69cb508d3b&mc_eid=5eda207752
and
https://github.com/mitchwongho/ESP-Alerts-for-Arduino/blob/master/sketches/esp32_android/esp32_android.ino
I have updated my espressif/esp32 with a pull and pulled in the BLE libraries (0.4.5)
I am on a linux mint 18.3 (64bit) machine, running Arduino 1.8.5 and tried with 1.8.4 just because.
and I am getting a lot of compile errors - I don't think it's the sketch.
I am pretty sure I missed something when I pulled and updated every - but have no idea what I might have missed.
My errors are here:
https://gist.github.com/kd8bxp/989d0cf3c84abbfdbee1882ed7333015

Any ideas what might be going on ? 
I'll try to answer any questions I would like to get this project working.
Thanks - LeRoy

chegewara, 2018-01-06:
Most likely you are missing ESP32_BLE library or the library is outdated.
https://github.com/espressif/arduino-esp32#installation-instructions

kd8bxp, 2018-01-06:
Thank you, I just pulled and updated everything last night (including git submodule update --init --recursive &&) which I think pulled in the latest BLE libraries.

chegewara, 2018-01-06:
`(.literal._Z14setupBLEServerv+0x14): undefined reference to `BLEDevice::createServer()'`
This usually means you have header files with declaration function but missing definition.

kd8bxp, 2018-01-06:
Ok, Thanks again - I kind of feel stupid - I have a library installed called BLEPeripheral (https://github.com/sandeepmistry/arduino-BLEPeripheral) I can't even remember why or what hardware I have that uses this - I removed that library and everything compiled and works.  I really feel dumb at this point making a "issue".  Thanks for the help.
-LeRoy

chegewara, 2018-01-06:
No problem, im glad you were able to find and fix it. You can close issue.

closed 2018-01-06 by kd8bxp