nkolban/esp32-snippets #348: BLE Library - new iBeacon advertising server data issue on iPhone

qrpfun, 2018-01-06:
I'm using Arduino IDE and I have imported only changes related to iBeacon by simply coping and overwriting BLEAdvertising.* BLEBeacon.* and BLEUUID.h inside Arduino libs.
Doing this way it compiles without any problem.

I used this code for server side iBeacons advertising:
https://pastebin.com/HYyGT5K2

Reading data with nRF Connect on iPhone it does not seems to be working as aspected.
After some info exchange we suspected the problem should be in iPhone data reading.
But may be there should be something else in arduino libs.

It should be interesting to understand where the issue can be using origina esp environment with an iPhone (do someone can tell me if I can install esp development environment without conflicts with the arduino one ?)

Here's raw data view in nRF Connect
[image]

chegewara, 2018-01-06:
From my point of view in this data packet is missing flags info. Im suspecting you have something messed with library. Can you try with this one:
https://github.com/nkolban/esp32-snippets/pull/315

qrpfun, 2018-01-07:
@chegewara thanks for the new suggestion. I did try to fully substitute Arduino IDE BLE folder with the one in your zip file. It all compiles as usual without problem, as mine, but final result is exactly the same as the one reported above. No change at all.
If it works for you, only difference remaining is the iPhone BLE stack

chegewara, 2018-01-07:
If you can try to resd with nRF connect on android phone. Its known issue with iPhones reading custom iBeacons.

qrpfun, 2018-01-07:
Ok, tomorrow I will get the esp32 with me at work and will ask a friend a try with an Android phone.
I will let you now.
Tnks

qrpfun, 2018-01-08:
I have just tried with an android phone and the beacon is recognized
without problem.
So, as you supposed, the problem is in the iphone.

chegewara, 2018-01-08:
@qrpfun Im just studying this documentation. It seems possible to build iBeacon that will be recognized by iPhone, but you have to tweak so many advertising parameters.
https://developer.apple.com/hardwaredrivers/BluetoothDesignGuidelines.pdf
I think so especially importand are advertising and/or connection parameters. See chapter 3, in particular 3.4 and 3.6.
Unlucky i dont have iPhone device so i cant perform test so if you have time and would like to make some test, please share with us results.

chegewara, 2018-01-08:
https://developer.apple.com/ibeacon/Getting-Started-with-iBeacon.pdf

qrpfun, 2018-01-08:
@chegewara thanks for the info.
I will try to take a look at that documentation when possible.
But does not seems so easy

gerryilagan, 2018-01-26:
@chegewara I'm trying to make iBeacon work on my ESP32 via Arduino but it seems that my copy of the examples doesn't have BLEBeacon.cpp/h -- how do I update the examples so that it will have the BLEBeacon files necessary to try out iBeacon on my ESP32?

chegewara, 2018-01-27:
@gerryilagan Hi, you need to update ESP32_BLE library or delete and install it again.

gerryilagan, 2018-01-28:
@chegewara it's on version 0.47 and couldn't find 0.49 (the version I see on GitHub) in the list of versions that I can install in Arduino

chegewara, 2018-01-28:
Just update arduino-esp32. You need to have installed git, then:
- go to \Documents\Arduino\hardware\espressif\esp32 on windows or ~/Arduino/hardware/espressif/esp32 on linux,
- update library with `git pull`
- git submodule update --init
- got to tools folder and run get.exe (windows) or python get.py (linux)
- have a fun

gerryilagan, 2018-01-28:
Hahaha... found what was missing in my update script.

Thanks @chegewara 👍 :-)

jrickard, 2018-02-14:
Just a suggestion, but iPhone requires iBeacons to transmit all advertising packets on a minimum interval of 100ms and a maximum interval of 100ms.  We are kind of hard coded, it appears to me, at 32ms min and 64ms max.

I don't know if it really matters.

chegewara, 2018-04-06:
Im looking for tester with iPhone to test ideas i have about iBeacon who is willing to contact on private (IRC or email).

jrickard, 2018-04-06:
I'm at jack@evtv.me.  I have iPhone and iPad.

And ESP32 of course.

qrpfun, 2018-04-06:
Let me know -I have those techs
Email giampiero.raschetti@gmail.com Skype giaras 

Inviato da iPhone

gerryilagan, 2018-04-07:
@chegewara - Do you still need help?

chegewara, 2018-04-07:
Sure, more tests = more ideas = better results.

gerryilagan, 2018-04-07:
What to do?

chegewara, 2018-04-07:
Ok, i will try to prepare bin file and post here asap.

chegewara, 2018-04-07:
Ita a first version, i hope its last ;) 

Because github doesnt allow to upload bin files ive added .txt, you know what to do.
[iBeacon_v1.bin.txt](https://github.com/nkolban/esp32-snippets/files/1886796/iBeacon_v1.bin.txt)

chegewara, 2018-04-09:
https://github.com/beaconinside/awesome-beacon
Can i ask someone to test this app with iPhone and tell me if iBeacon created with it can be recognized?
https://play.google.com/store/apps/details?id=net.alea.beaconsimulator