nkolban/esp32-snippets #351: [Question] Dynamically Start Restart Bluetooth (BLE)

mirko314, 2018-01-07:
I am trying to Stop Bluetooth after some time but did not find any fitting function in the c++ snippets.
Using the following lines i can successfully reduce current consumption, so it does probably work for shutting BT down. 
```
void stopBLE(){
  Serial.println(F("Shutting BT Down ..."));
  Serial.print("esp_bt_controller_disable");
  esp_bt_controller_disable() ;
  Serial.print("esp_bt_controller_deinit ");
  esp_bt_controller_deinit();
  Serial.print("BT Shutdown finished");
}
```
Trying to restart it with snippets like
`BLEDevice::init("ESP");`
Does not work properly. 
Has somebody considered implementing a shutdown / reinit function for Bluetooth?
Awesome Library!

pit001, 2018-03-01:
Would be also interesting by me.

rweller59, 2018-04-01:
I'm working on patterns to reduce power and working on shutting down BLE but not putting chip to sleep. I am successfully shutting down bt radio I believe but struggling to bring back up the ble services.

I'm wondering what the suggested steps would be. I'm using ble server, callback and advertising.

I've tried setting pServer and pService to nullptr in shutdown but when I try to bring services back up I hang somewhere with the console reading

bangonkali, 2019-02-19:
Any news on this? I have encountered the same problem.

JJens, 2019-07-04:
Did anyone solve this issue? I got the same challenge...

istoryahi, 2019-10-20:
My solution was a bit inelegant but here it goes. 

The problem: I need to switch between WiFi and Bluetooth dynamically.

Solution: During a switch, I store all relevant information to Flash and the target state either BLE Mode or Wifi Mode. Restart the Device. Read the target state, initialize new Mode. I do this from BLE -> WiFi and WiFi to BLE.

chegewara, 2019-10-20:
I dont have any problem with arduino v1.0.3 and BLE library. I can init and deinit BLE on demand any time. The problem can be with all GATT attributes, because you need to control how they are created and deleted.

istoryahi, 2019-10-20:
> I dont have any problem with arduino v1.0.3 and BLE library. I can init and deinit BLE on demand any time. The problem can be with all GATT attributes, because you need to control how they are created and deleted.

Have you had any success switching between WiFi and BLE during runtime, @chegewara ?

chegewara, 2019-10-20:
Yes. Like i said, currently i am working on arduino as component in platformio environment project. In project there is wifi + AWS IoT + BLE. I am init and deinit BLE on demand, mostly BLE is used to wifi credentials provisioning and deinit on wifi connection, then get back init on wifi disconnect.
I dont remember what procedure has been used in other arduino project, but in that one BLE is used to print on BLE POS thermal printer and wifi for OTA update in STA or AP mode (http client or server).

istoryahi, 2019-10-20:
@chegewara Awesome! It would be nice if you could share some code snippets on how exactly you init/deinit BLE and switch to Wifi Mode on-demand without restarting.

chegewara, 2019-10-20:
There is not much to share:
`BLEDevice::deinit(false);` <--- if true, then free BLE memory and can no longer use BLE
Then delete all attributes with `delete pServer' etc or you have memory leak. Some attributes are deleted by this library, so you can encounter some problems. (i am struggling with this currently, which attributes i should to delete).

To init, use the same procedure like first time (usually in setup()), because this library will remember all GATT attributes, but low level library (esp-idf) will forget all on de-init.

mauroscianca98, 2019-10-20:
@chegewara that's awesome, but it would be even more if you could share with us some code! api and docs do not tell a single thing about this situation

chegewara, 2019-10-20:
I dont understand what else you want me to share?
To deinit BLE just use this line:
`BLEDevice::deinit(false)`

To init BLE use code similar to this:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino#L119-L129
or this:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino#L21-L38

istoryahi, 2019-10-20:
thanks @chegewara . I will try and make a simple routine for switching between Wifi and BLE. Event using deinit for both Wifi and BLE, my last attempt seemed to have ended in failure my apologies. 

I will try this again.

istoryahi, 2019-10-20:
> @chegewara that's awesome, but it would be even more if you could share with us some code! api and docs do not tell a single thing about this situation

It's inelegant but try restarting between switching modes from BLE, WIFI UPDATE, WIFI CLIENT, WIFI SERVER. You may need to persist relevant information to Flash before restarts though. And make sure to have an external RTC module taking care of time so you do not lose track of time between restarts. 

Restart routine is pretty quick though, in my application, this is not done very often. Only in cases where the WIFI Credentials are being set up the user needs to be in BLE MODE, once it's set up, it will be in Wifi mode for the rest of the lifetime of the device unless the Wifi router is changed.