nkolban/esp32-snippets #352: BLE ASSERT_PARAM crash when scanning takes too long

ThePizzaguy, 2018-01-08:
Hi Kolban,

Recently working with your code on a project I get this error if i use my code in the onResult function from your BLEScan. 
It only happens when my code takes a certain amount of time because if I use alot simpler code it works fine for hours.
[image]

I did see your post on the ESP32 forums about this from a month ago and were wondering if you found a fix.

Best Regards

Oscar

chegewara, 2018-01-08:
If you are working on esp-idf then you can try one thing. In menuconfig find watchdog time and try to increase it. Im not saying it will help, but from screenshot we can see it may be related.

ThePizzaguy, 2018-01-08:
Yes i am using esp-idf and that gave some interesting results, the code lasted somewhat longer and the error message was longer as well but sadly it didn't work. I found that the length of the error message corresponds with the time it takes in the function.

[image]