nkolban/esp32-snippets #353: BLE indicate gets stuck on disconnect

tijnkooijmans, 2018-01-08:
When using BLE indicate the task often gets stuck on a BLE disconnect. This is because the semaphore that keeps waiting. A call to 'm_semaphoreConfEvt.give();' on disconnects would solve this.

nkolban, 2018-01-08:
Howdy my friend.  Would you be able to provide a trace output showing the puzzle?  

I'm guessing that this is the implementation of the disconnect() method we are talking about.

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L130

What I'd like to see if the path/state where we lock up.

tijnkooijmans, 2018-01-09:
The flow to get this error is as follows:

- Send indication on a characteristic (this calls wait on m_semaphoreConfEvt)
- Before the client can reply to the indication, disconnect.
- Now the indication is stuck waiting. 

To resolve this problem 'm_semaphoreConfEvt.give();' should be called to the characteristic on disconnect events.

As a work-around I added a disconnect function to the characteristic that I call external to your library on disconnect events: https://github.com/StudioSophisti/ESP32_BLE_Arduino/commit/eb46789e978c58a285891e14353f7c01ff5a8593

The proper solution would be to handle this internally of course.

nkolban, 2018-01-09:
Howdy my friend.  I had a look through the source code.  Let me replay the story to you to make sure I have it correct.

You are writing an ESP32 BLE Server.  You have created a service with characteristics.  In your code you are performing a `BLECharacteristic#indicate()` call which will will send an indication to any connected BLE clients.  We see this code here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L471

Since the indicate wants to wait for a positive indication, it blocks here waiting for the response:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L512

However, before an indicate response arrives, the client disconnects and the response will NEVER arrive and hence the code locks up.   Is that the thinking?

From my initial examination of the BLE code, I see that when a client disconnects, the ESP32 will receive a disconnection event.  That event "should" be propagated down to all the services and characteristics.  See for example:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L446

If I am reading this correctly, then a disconnection request from a client "should" release the lock that the previous `indicate()` was waiting upon.  I'm not able to eye ball a problem (naturally this does NOT mean that there isn't a problem).

It looks like Mr @Sepp62 did some work on this area back for issue #209.

Two things thus occur to me:

1) Lets see if we can't determine what version of the code your are running ... it may be prior to #209 in which case there would be an answer.

2) Can you capture and send the debug trace.  That will show clearly what path is being executed and whether or not we are performing the unlock.

Sepp62, 2018-01-10:
Yes, I indeed made changes in this area. But these were done for "notify". The semaphore for "indicate" has been there before (that's were I got the "inspiriation" from).

I also agree with Neil, that disconnect frees the semaphore.

chegewara, 2018-11-15:
Should be fixed in master.

closed 2018-11-15 by chegewara