nkolban/esp32-snippets #354: BLE deep sleep

athankok, 2018-01-09:
I am using the notify example. I want to deep sleep the ESP32 but it seems like it never goes to deep sleep (current consumption 60 ma). Is there a way to deinitialize the ble server ?
esp_bt_controller_disable();
delay(1000);  
esp_sleep_enable_ext0_wakeup(GPIO_NUM_0,0);

nkolban, 2018-01-10:
You ask a good question ... but it is one we haven't studied before.  We have primarily been focusing on the ESP-IDF samples to get bluetooth running ... and not yet to shut it down.  There appears to be some documentation here:

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_bt_main.html

it seems like we have APIs such as:

* esp_err_t esp_bluedroid_deinit(void)
* esp_err_t esp_bluedroid_disable(void)

which may need some invocation.  What I'd suggest is posting a question to the ESP32 SDK forum and see if there are recommended practices for "shutting down" the BLE subsystem cleanly.  If any of these need to be integrated with the C++ BLE library, I'll be happy to add those.

chegewara, 2018-01-11:
http://esp-idf.readthedocs.io/en/v3.0-rc1/api-reference/system/sleep_modes.html#wifi-bt-and-sleep-modes

chegewara, 2018-01-11:
https://esp32.com/viewtopic.php?f=2&t=4185#p18802

athankok, 2018-01-11:
Dear all thanks for the help ,
I am finally able to sleep the esp32 .
 esp_sleep_enable_ext0_wakeup(GPIO_NUM_0,0);
 esp_bluedroid_disable();
 esp_bluedroid_deinit();
  btStop();
 delay(1000);
 esp_deep_sleep_start();

athankok, 2018-05-17:
Yes i can confirm it.I did it with the following : 
esp_bluedroid_disable(); 
esp_bluedroid_deinit();
You need first to disable and then to free the resources thats why you deinit .

fgaultier, 2018-05-18:
Currently I'm not even initializing Bluedroid, just enabling the BLE controller and then disabling it before going to deep sleep. Everything works fine except that the consumption stays high during deep sleep: 60 mA instead of 1-2mA (this is what I get if do not enable the BLE controller at all).  What kind of hardware platform do you use ?

athankok, 2018-05-18:
Not sure if i am wrong but there is a similar function for the bluetooth controller  esp_bt_controller_deinit().
I have test it on ESP32-DevKitC V4 with ESP-WROOM-32

pdthang, 2019-11-30:
> Not sure if i am wrong but there is a similar function for the bluetooth controller esp_bt_controller_deinit().
> I have test it on ESP32-DevKitC V4 with ESP-WROOM-32

Can you share your BLE Notify code in combination with deep sleep?
Thank you so much.

Nisar-ali, 2020-06-05:
can you share your code