nkolban/esp32-snippets #355: HttpServer listen socket needs to do reuse_addr

yaoyogi, 2018-01-09:
When the HttpServer is started ,and takes a request, and then is stopped, you cannot restart the server since you will get a socket bind error saying the port is already bound.

The problem is, even though the listen socket is closed, it takes LWIP awhile for the underlying connection to actually relinquish the port.  By having the HttpServer listener socket do reuse_addr, we can avoid this problem

nkolban, 2018-01-10:
Howdy my friend.
It is my belief that the reuse of sockets is enabled at the "make menuconfig" level as opposed to in code at the HttpServer library level.   Have you tried switching on this make menuconfig option and rebuilding the code to see if there are any changes?

[image]

yaoyogi, 2018-01-10:
I have this set for this and other projects for similar reasons.  However, I have had to set the reuse addr option on the socket itself for it to work.  I interpreted this option as its possible to do the reuse addr vs. it's not possible; compared to all sockets are reuse_addr.

I have latest IDF and cpp_utils (as of 1/9/18).   The scenario is:
1. Create HttpServer and start the server
2. Process a request
3. stop the HttpServer
4. start the HttpServer again
5. Any incoming request will fail on the listener socket saying the address is already bound/in-use.

When I change the HttpServer's listener socket to reuse_addr the problem is fixed.

For the above scenario you don't run into this problem?

I have other socket-server code where I hit this problem before and just setting the menu config was not sufficient.

nkolban, 2018-01-10:
Testing now ... will be back to update this post shortly....

nkolban, 2018-01-10:
Thank you sir.  After I tested following your recipe I was able to fully recreate.  The puzzle has now been resolved with some code fixes that have now been merged into the repository.  If you could pull/clone and see if that meets the needs, that would be great.

yaoyogi, 2018-01-10:
Pulled and verified it fixes the reuse_addr issue.  Thank you very much!

closed 2018-01-10 by yaoyogi