nkolban/esp32-snippets #357: _WindowOverflow4 during startup

squonk11, 2018-01-11:
My current SW compiles o.k. but during startup of the µC I receive strange messages (in yellow): 

> 0x40080000: _WindowOverflow4 at E:/msys32/opt/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

and 

> 0x400d0018: _flash_cache_start at ??:?

and

> 0x4008bfa4: bb_init at ??:?

The whole startup sequence log is: 

>                        I (30) boot: ESP-IDF v3.0-dev-1517-g8051ac46-dirty 2nd st
age bootloader
I (30) boot: compile time 19:27:47
I (30) boot: Enabling RNG early entropy source...
I (31) boot: SPI Speed      : 80MHz
I (322) boot: SPI Mode       : DIO
I (32) boot: SPI Flash Size : 4MB
I (33) boot: Partition Table:
I (33) boot: ## Label            Usage          Type ST Offset   Length
I (34) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (35) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (36) boot:  2 factory          factory app      00 00 00010000 00200000
I (37) boot:  3 storage          Unknown data     01 81 00300000 00100000
I (38) boot: End of partition table
I (38) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x20c28 (134
184) map
I (58) esp_image: segment 1: paddr=0x00030c50 vaddr=0x3ffc0000 size=0x0340c ( 13
324) load
I (60) esp_image: segment 2: paddr=0x00034064 vaddr=0x40080000 size=0x00400 (  1
024) load
0x40080000: _WindowOverflow4 at E:/msys32/opt/esp/esp-idf/components/freertos/xt
ensa_vectors.S:1685

I (60) esp_image: segment 3: paddr=0x0003446c vaddr=0x40080400 size=0x0bba4 ( 48
036) load
I (68) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8f6e0 (587
488) map
0x400d0018: _flash_cache_start at ??:?

I (150) esp_image: segment 5: paddr=0x000cf700 vaddr=0x4008bfa4 size=0x04760 ( 1
8272) load
0x4008bfa4: bb_init at ??:?

I (153) esp_image: segment 6: paddr=0x000d3e68 vaddr=0x400c0000 size=0x00000 (
   0) load
I (156) boot: Loaded app from partition at offset 0x10000
I (156) boot: Disabling RNG early entropy source...
I (156) cpu_start: Pro cpu up.
I (157) cpu_start: Starting app cpu, entry point is 0x40081010
0x40081010: call_start_cpu1 at E:/msys32/opt/esp/esp-idf/components/esp32/cpu_st
art.c:222

I (0) cpu_start: App cpu up.
I (158) heap_init: Initializing. RAM available for dynamic alloocation:
D (159) heap_init: New heap initialised at 0x3ffaff10
I (159) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (160) heap_init: New heap initialised at 0x3ffcb4f0
I (161) heap_init: At 3FFCB4F0 len 00014B10 (82 KiB): DRAM
I (162) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (162) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (163) heap_init: New heap initialised at 0x40090704
I (164) heap_init: At 40090704 len 0000F8FC (62 KiB): IRAM
I (165) cpu_start: Pro cpu start user code
D (172) clk: RTC_SLOW_CLK calibration value: 3046746
D (24) intr_alloc: Connected srcc 46 to int 2 (cpu 0)
D (24) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (26) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (27) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (28) heap_init: New heap initialised at 0x3ffe0440
D (29) heap_init: New heap initialised at 0x3ffe4350
D (29) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (87)) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (2551) main: -->main
D (2553) intr_alloc: Connected src 37 to int 13 (cpu 0)
Name: SU08G
Type: SDHC/SDXC
Speed: default speed
Size: luMB
CSD: ver==1, sector_size=512, capacity=15523840 read_bl_len=9
SCR: sd_spec=2, bus_width=5
I (2631) main: Opening file
I (2633) main: File opened an closed
D (2633) intr_alloc: Connected src 36 to int 17 (cpu 0)
I (2634) main: IDF Version: v3.0-dev-1517-g8051ac46-dirty; Free Heap: 67360 byte
s

chegewara, 2018-01-11:
`Size: luMB` ???

squonk11, 2018-01-11:
That is the output of `sdmmc_card_print_info(stdout, card);`

nkolban, 2018-01-11:
I am assuming that you were able to compile and flash your app in the past and now you are no longer able to do the same task.   That usually points to something having changed.  Did you do anything new/different recently?

Another old favorite ... is that when things suddenly stop working for no reason is to erase the flash using esptool.   This returns the ESP32 to its factory default.  On occassion, especially during development, we can corrupt flash storage or power off at "just the wrong time" and a reset of flash never does any harm.

chegewara, 2018-01-11:
Ok, but how much is 'lu' MB? Even if change this to hex value it will be about 27765 or 108117.
BTW is it all that is printed in serial log?

squonk11, 2018-01-11:
Now I erased the flash using `make erase_flash`. After that I had to enter the access point credentials in the bootWifi page. After that the ESP boots again. But unfortunately the error message persists.

squonk11, 2018-01-11:
@chegewara: The size of the SD card is 8GB. I remember that in the past this value was detected and shown correctly.

chegewara, 2018-01-11:
When im looking at this logs i think it should be posted on esp-idf board. It is related to memory mapping and issue exists even before app_main() starts.

squonk11, 2018-01-11:
recently I changes some parameters in menuconfig (in order to find a clue for #344 ). I will not try to figure out which values I changed and set them back to original settings. Maybe I'll find the reason.
Yes, the message appears during the startup of the ESP. So I think it is probably related to some settings in menuconfig.

squonk11, 2018-01-11:
I found a similar topic here: [https://github.com/espressif/esp-idf/issues/447](url). Here they say that this message is not harmful (if my topic is really the same...). I think I will ask at Espressif forum.

squonk11, 2018-01-13:
I received an answer from teh esp32 forum: [https://www.esp32.com/viewtopic.php?f=13&t=4199](url) According to this these messages are not harmful. We can close this issue.

closed 2018-01-13 by squonk11

rustyx, 2019-04-09:
The link is wrong. Here's a working link: https://www.esp32.com/viewtopic.php?f=13&t=4199