nkolban/esp32-snippets #363: Latest esp-idf breaks cpp_utils/WiFi

rweller59, 2018-01-17:
Apparently m_mdns_server is no longer defined in last esp-idf msdn.h

nkolban, 2018-01-17:
Thank you sir.  Trivially recreated ... looking for solution now.

nkolban, 2018-01-18:
Until we can make progress on the issue of MDNS ... I have commented it out.  This will, of course, remove MDNS functions from the solution but I feel that they might not be as heavily used as the other components.  For the majority of users, I'm hoping there will be a clean compile.  Do you yourself need MDNS?  If not, do you now get a clean compile with the latest code downloaded?

rweller59, 2018-01-18:
thanks Neil. no, I do not need MDNS. I started to comment everything out as well today but ended up going back to previous version of esp-idf. I will update and try again tomorrow.

chegewara, 2018-01-18:
I think its time to create new branch for v3.0 because esp-idf master is now v3.1.

nkolban, 2018-01-18:
The master being a 3.1 early draft is not the same as there being a 3.0 release.

If we look here,  https://github.com/espressif/esp-idf/releases

We don't yet (as of this day) see a 3.0 release.   When that occurs, we will create a branch that will be cpp_utils against THAT release and a master branch that will always be against master branch of ESP-IDF.  However, for a V3.0 based cpp_utils, there has to be an ESP-IDF that we all agree is v3.0.

nkolban, 2018-01-27:
Aha ... some more of the mystery being resolved ...

https://github.com/espressif/esp-idf/issues/1545

snahmad, 2018-03-10:
Hi,
I am using c++ Wifi class to connect to my Access point. i am specfing static ip address. It is working. how can i use DHCP to get dynamic ip address. how can i set name to for my device. does c+ support these operations or i can use C functions.
Thanks,
Naeem