nkolban/esp32-snippets #367: Enhance Task class

squonk11, 2018-01-18:
I would like to suggest to enhance the Task class with the possibility to select the core on which the task has to run. This could be done in the following way:

Declare constructor for class Task (in Task.h): 
`Task(std::string taskName="Task", uint16_t stackSize=10000, uint8_t priority=5, BaseType_t core_id=tskNO_AFFINITY);`

define variable for the core id (in Task.h): `BaseType_t  m_core_id;`

Define constructor for class Task (in Task.cpp): 
`Task::Task(std::string taskName, uint16_t stackSize, uint8_t priority,  BaseType_t core_id) {
	m_taskName  = taskName;
	m_stackSize = stackSize;
	m_priority  = priority;
	m_taskData  = nullptr;
	m_handle    = nullptr;
	m_core_id	= core_id;
} // Task `

And invoke the task using the core id (in Task.cpp): 
`::xTaskCreatePinnedToCore(&runTask, m_taskName.c_str(), m_stackSize, this, m_priority, &m_handle, m_core_id); `

Is it possible to add this functionality?

Best regards

nkolban, 2018-01-18:
I don't see why not ... give some thought to the parameters.  Do we want to pass "ALL" parameters on the constructor ... or do we want to pass minimal parameters on constructor and provide "setters" than can be called before "start"?

For example:

```
MyTask t = new Task(...);
t.setCore(CPU0);
t.start();
```

squonk11, 2018-01-19:
Will you implement this functionality or shall I provide it?

nkolban, 2018-01-19:
Howdy my friend ... do you have a preference?  I like to code .. .but maybe you do to and you actually want to provide it?  I'll bow to your preference.   If you want me to do it, this weekend is about the soonest I can look at it.

squonk11, 2018-01-19:
I can do it this evening or tomorrow. 
How do I send you the new files? Just by eMail? Or should I use a smarter way over github - but this I have never done before; so I would need a hint how this can be done via github.

nkolban, 2018-01-19:
Learn how to use IRC chat rooms and find the `#ESP32` room.  I'm always in there as "Kolban".  You can ping me directly there.  I'll leave it up to you on how to make code changes ... but there is a "correct" method.

It involves learning enough about Github to be dangerous.

At the highest level:

1) You fork the cpp_utils repository.  This will create your own private copy.
2) You clone your copy of the forked repository and make source code edits on your PC.
3) You commit your changes back to your copy of the forked repository.
4) You submit a "Pull Request" on your copy of the forked repository which informs the master cpp_utils repository of the changes requested to be merged in
5) I "click the button" and the changes are merged in.

squonk11, 2018-01-19:
That sounds interesting. I will try it tomorrow.

squonk11, 2018-01-20:
I now submitted a Pull Request. Can you check if it is o.k.?

nkolban, 2018-01-20:
That looked absolutely perfect to my eyes.  Comments and everything.   The process worked perfectly.  At my end I simply had to eyeball the code and then click "merge" and it was done.  You worked the process perfectly!!!   How did you find it?  Was it straightforward enough?  Whatever you did ... you did it perfectly from where I sit.  You changes are now permanently part of the library.  Thank you kind sir and welcome to the team.

squonk11, 2018-01-20:
I had to read a little bit about the git commands. I found that with `git commit -a` the files were not updated in the web. After searching for the problem I found that additionally I had to do a `git push` also. The rest was straightforward.

closed 2018-01-23 by squonk11