nkolban/esp32-snippets #368: ESP_BLE_Arduino 0.4.8 & BLEUuidNotFoundException

reaper7, 2018-01-18:
I update ESP_BLE_Arduino to 0.4.8 and copy HIDTypes.h as **chegewara** wrote:
https://github.com/nkolban/esp32-snippets/issues/365

but I got this error message:
```
C:\PROGRAMY\arduino_projekty\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':

C:\PROGRAMY\arduino_projekty\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:336:12: error: exception handling disabled, use -fexceptions to enable

  throw new BLEUuidNotFoundException;
```
where I must use -fexceptions ??

P.S. arduino-esp32 core update to latest git (with xtensa-esp32-elf-win32-1.22.0-80)

nkolban, 2018-01-18:
Unfortunately I have a day job (9-5 US time) and Ive also picked up an ESP32 consulting project (yeah) that is eating most of my spare time.  However, that said ... it is abhorrent to me to leave users in the lurch who have started to build on libraries that I have put together.  New functions are likely on hold until I can come up for air but breakage to existing function is unconscionable.  I got a ping the other day from mr @me-no-dev who is our Arduino-ESP32 owner to tell me that it was time to re-build BLE for the Arduino library.  I packaged up the best so far current BLE and *hoped* I'd gotten it right ... but it appears likely not.  This evening I will **make** time to test and make sure we have a clean compile and run and ... if not, either fix or _prod_ others.   .... to be continued.

me-no-dev, 2018-01-18:
you are missing some headers (and maybe sources) in the Arduino lib. Not a biggy :)

nkolban, 2018-01-19:
I made the changes to the repo by adding the missing files

https://github.com/nkolban/ESP32_BLE_Arduino

I can see the new files in the repo.  However, when I follow my instructions for building an Arduino environment:

```
$ mkdir espressif
$ git clone https://github.com/espressif/arduino-esp32.git esp32
$ cd esp32
$ git submodule update --init
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into '/home/kolban/bin/arduino-1.8.5/hardware/espressif/esp32/libraries/BLE'...
Submodule path 'libraries/BLE': checked out '3f5005c383540d07dd16ac044c78d7be28d4aa8e'
```
I get the message that says the "older" commit is still being pulled (i.e. the one that doesn't have the files).

I've deleted my local arduino directory but still no change when I recreate.  Its almost as though my local environment has somehow cached what it believes to the be Ardunio BLE repo.

Paging mr @me-no-dev to see if he has a notion.

me-no-dev, 2018-01-19:
lib updated and all fixed :) no more build errors

reaper7, 2018-01-19:
👍 after update arduino-esp32 and ble lib submodule everything works fine.
time to test client onDisconnect callback

tnx!

closed 2018-01-19 by reaper7

ifrew, 2018-03-05:
I updated my esp32 core today and am getting the same issue, ie calls to throw exception are asking for -fexceptions to be enabled. I deleted my existing esp32 directory and cloned from scratch, did the recursive git update, cleaned my solution in visual studio and rebuilt Have done this a number of times now, Do I have to pull down manually pull down the BLE ardunio library also from this repository? Reading the above thread it seems that its was integrated into the core branch. Any assistance appreciated. Trying to see if the there is more free heap now..

ifrew, 2018-03-07:
Its ok I found the problem. The latest core I downloaded has in the platform.txt file both -fno-exceptions and -fexceptions set for the cpp compiler! There is a note in GitHub about that change since January but not fixed yet. I juest deleted the -fno-exceptions and restarted the IDE and it compiles now.

reaper7, 2018-03-07:
indeed! -fno-exceptions and -fexceptions together...
Please, try to link this issue to the "note" on arduino-esp32 repo,
maybe someone will be interested.