nkolban/esp32-snippets #369: LibCurl RESTClient memory consumption

rweller59, 2018-01-18:
Been trying to launch curl restclient example and get libcurl error 27 (CURLE_OUT_OF_MEMORY). I continue to get this error until I set launch task stacksize to 70000. (at this point I get protocol https not supported).

Is this a potential issue in cpp_utils? or just the facts with libcurl. If so, makes it pretty much impossible to use ble + wifi along with any libcurl functions.

rweller59, 2018-01-19:
From what I can see, there is not much to be done about this memory consumption. I am abandoning libcurl and moving to rest calls with mbedtls. In my first test, bin size has shrunk dramatically and heap consumption is done about 50% from what libcurl used. I don't have much experience with this stuff - hopefully I'm not losing much functionality going this route.

mgleason, 2018-01-22:
So, a few things:

"> ...error until I set launch task stacksize to 70000"
Have you tried running the sample test (cpp_utils/tests/test_rest.cpp)?
...it's stacksize is 12000.

"> (at this point I get protocol https not supported)"
Out of curiosity, where did you see that?

From what I can see, restClient doesn't care one way or another whether https is supported.  If you call restClient.setURL(url) with a url that starts with `https://`, restClient will use the exact same url when it configures curl (using `curl_easy_setopt(m_curlHandle, CURLOPT_URL, m_url.c_str())`).

So, if you're getting an error message like:
> Protocol "https" not supported or disabled in libcurl

It's because libcurl isn't configured to support https.  So, it's up to you to configure libcurl to support https.  I use mbedtls.  I don't remember having to download it - but there may have been some settings in "make menuconfig".  The main thing I did was edit `curl/curl_config.h` as so: 

`#define USE_MBEDTLS 1`

"> From what I can see, there is not much to be done about this memory consumption."
Unless you're firing off a bunch of requests in parallel, it's not clear why you'd need so much stack space.  

"> I am abandoning libcurl and moving to rest calls with mbedtls."
That's certainly a option!  In fact, I actually tried this - thinking I'd save a bunch of memory.  But, I'm lazy and like using well tested libraries.  Plus, once I got past doing simple POST's, I realized that curl was doing a lot for me and I was reinventing the wheel, adding and testing other libraries and code to build the same thing.  I also found that my initial CURL test code had some problems that was causing it to use a lot more memory than needed.

So, fixed the problems, gave curl another try and it's been fine.  Adding TLS/SSL (mbedtls) later did increase the stack size some.  But nothing like 70000.  Right now, with fairly simple (and serialized) requests, it runs fine with a stack size of 10240.

"> In my first test, bin size has shrunk dramatically and heap consumption is done about 50% from what libcurl used."

Wow - if you can write a generalized restClient that uses a ton less heap/stack than curl, that would be great!  I'll use it!!!

But unless your requirements are quite minimal, it might be challenging.  I mean networking can use a LOT of memory, right?  Buffers for requests, buffers for responses... And you're still going to be using mbedtls which needs memory of it's own - e.g. doesn't mbedtls's default settings allocate a 16K buffer?!?

Although there are some libraries out on GitHub that might do just that:

- C [HISONA https_client ](https://github.com/HISONA/https_client)
- C++ [jgaa restc](https://github.com/jgaa/restc-cpp)
- This one raps curl (like this library!) [mrtazz restclient](https://github.com/mrtazz/restclient-cpp)

If you find one that's great I'd love to know!

Jumping back to curl/mbedtls - One note here with respect to the binary, once I added both curl and mbedtls, the binary _was_ significantly bigger and no longer fit in the default 1m partition.   I ended up increasing the partition size to 1.5M - See [Partition Tables-Creating Custom Tables](http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html#creating-custom-tables)

And, of course another option is to decrease/optimize the memory used by curl/mbedtls:

1. mbedtls - [reduce memory footprint](https://tls.mbed.org/kb/how-to/reduce-mbedtls-memory-and-storage-footprint) gives a number of suggestions.   The only one I've tried so far is decreasing MBEDTLS_SSL_MAX_CONTENT_LEN to 8192 (though I should be able to go a lot smaller).
2. curl - [See "Reducing Size"](https://curl.haxx.se/docs/install.html) gives a number of options for reducing the size of the binary.  

It would be terrific if someone could publish a set of options to use - for folks like us who just want to call RESTful api's and don't need all the other wonderful things curl does (like FTP, SMPT, TELNET, etc...)

rweller59, 2018-01-22:
I started with the RestClient example so it was a pretty simple test. At first RestClient logged an error that just printed garbage to the screen - went in and checked the error code (27) which is when I started increasing stacksize. Error finally went away when I got to 70,000 and error message then became readable. So next error was the https not supported.

I had downloaded latest version of libcurl and could not find any library configuration settings to 'enable' https. I assume that the https issue could eventually be resolved but everything I found googling about libcurl memory requirements seemed to indicate that is is a hog.

I agree that using an established library is preferable and I'd much rather go that way but at the moment my needs are minimal. I am concerned that I'll eventually want more of the functionality that libcurl provides and maybe I'm just kicking the bucket down the road a bit.

mgleason, 2018-01-22:
Whichever you choose, good luck! 

A couple more notes in case you decide to go back to curl/mbedtls (or someone else finds this issue):

- Weird you had so much trouble with the restClient sample test (cpp_utils/tests/test_rest.cpp).  Last time I tried it, it worked fine unmodified.  When saying that curl is a memory hog, I do think it's important to understand which types are affected and how.  
Binary - I'll definitely agree that curl and mbedtls significantly increase the size of the binary.  You can increase the program size (see above).  This may be fine, but it you need multiple images, eg OTA, it could be a problem.  
Stack - I've had to increase it by maybe 4,000 (e.g. setStackSize(12000)) but nothing like the 70,000 you experienced.  
Heap - mbedtls's default will suck up 16K - so you may want to change that (see above).  Haven't looked at curl so can't say.

- If you're trying to install curl on a Mac OS/X:
[these notes might help](https://github.com/nkolban/esp32-snippets/issues/51#issuecomment-359530630)

Specifically to "could not find any library configuration settings to 'enable' https": 
1. The [curl install doc](https://curl.haxx.se/docs/install.html) lists options for configuring SSL and TLS:  see:  "./configure --with-ssl", "--without-ssl --with-mbedtls", etc.
2. Although, as I mentioned above, all I did was edit curl/curl_config.h as so `#define USE_MBEDTLS 1 `. 
3. And, now that I think about it, running `configure --without-ssl --with-mbedtls` may do the same thing (as editing curl/curl_config.h).

closed 2018-03-12 by rweller59

snahmad, 2018-05-18:
Does LibCurl with https client works on esp32 within 512 internal RAM?