nkolban/esp32-snippets #370: BLE Server example fails to connect

JasXSL, 2018-01-19:
What I've tried:

1. Arduino IDE. File > Examples > ESP32 BLE Arduino > BLE Server
2. Compile and upload.
3. Output from COM:
<pre>
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
Starting BLE work!
Characteristic defined! Now you can read it in your phone!
</pre>
4. It shows up in my phone. I click connect and it tries for a while but then goes into Connection Fail. No additional output shows up on COM.

Does this mean the hardware is faulty? Is there a way to debug this?

chegewara, 2018-01-19:
Please change debug level to verbose:
Tools->Core debug level->Verbose

JasXSL, 2018-01-19:
Now I get the following:

<pre>
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
Starting BLE work!
[D][BLEDevice.cpp:58] createServer(): >> createServer
[D][BLEServer.cpp:297] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1755] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:301] registerApp(): << registerApp
[D][BLEDevice.cpp:61] createServer(): << createServer
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914c
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1680] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914c, inst_id: 0]]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:160] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914c)
[D][BLEService.cpp:166] setHandle(): << setHandle
[D][BLEService.cpp:89] executeCreate(): << executeCreate
[D][BLEServer.cpp:93] createService(): << createService
[D][BLEService.cpp:188] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:191] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26a9 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914c, handle: 0x28
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEService.cpp:203] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:629] setValue(): >> setValue: length=21, data=48656c6c6f20576f726c642073617973204e65696c, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a9
[D][BLECharacteristic.cpp:636] setValue(): << setValue
[D][BLEService.cpp:125] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914c, handle: 0x28
[D][BLECharacteristic.cpp:82] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:93] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: beb5483e-36e1-4688-b7f5-ea07361b26a9, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914c, handle: 0x28
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1633] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a9]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:574] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=beb5483e-36e1-4688-b7f5-ea07361b26a9
[D][BLECharacteristic.cpp:576] setHandle(): << setHandle
[D][BLECharacteristic.cpp:134] executeCreate(): << executeCreate
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1769] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 0x28]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEService.cpp:151] start(): << start()
[D][BLEAdvertising.cpp:86] start(): >> start
[D][BLEAdvertising.cpp:105] start(): - no services advertised
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:137] start(): << start
[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
Characteristic defined! Now you can read it in your phone!
[D][BLEUtils.cpp:1075] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1261] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1067] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
</pre>

Nothing shows up in the log when I try to connect. I don't think the phone is the issue because I have no problems connecting to a raspberry pi with it.

chegewara, 2018-01-19:
Everything looks normal from this log. BLE server starts and is advertising.

JasXSL, 2018-01-19:
Could it be a version issue? My phone says it supports bluetooth 4.1, and the ESP says 4.2?

chegewara, 2018-01-19:
BLE 4.2 is backward compatible, my android phone is v4.0 and i dont have issues. Maybe its something with new BLE library, but like i said, from this log i dont see any problems. Maybe try with nRF connect and look at logs in it?

JasXSL, 2018-01-19:
nRF connect worked, but none of the other bluetooth testing apps I used. Weird. I'll close this for now.

closed 2018-01-19 by JasXSL