nkolban/esp32-snippets #372: Cannot detect/scan classic bluetooth devices

Lakoja, 2018-01-21:
I acquired an ESP32 recently (a ESP-WROOM32-Breakout).
And tried the ble_scan example with Arduino. This works ok: Upload works. Some BT devices are listed as found.

However the devices listed there are only "low energy" ones.

This is true even if I define _CLASSIC_BT_ENABLED_. And as another test I even set the _ESP_BT_MODE_CLASSIC_BT_ for _esp_bt_controller_enable()_ in BLEDevice.cpp.

Parallely I checked this with an app on my Android phone ("BlueScan") which also shows some devices as type "Classic". All of which are not found by the ESP32.
(The one I am actually looking for also shows up a classic there...)

Does this sound like a hardware defect? Or are there any shortcomings in the libraries known?

chegewara, 2018-01-21:
As you can see this is BLE library which means it only works with bluetooth LE. You cant work with bluetooth classic devices or use ESP32 as bluetooth classic device.

Lakoja, 2018-01-21:
Is there a purpose for the _CLASSIC_BT_ENABLED_ flag from BLEDevice.cpp?

chegewara, 2018-01-21:
No and yes. When memory release will be implemented eventualy it could crash user application if it will be dual mode app.

Lakoja, 2018-01-21:
But the underlying sdk supports classic bluetooth? Because the ESP32 does according datasheets.

Thank you for your time and patience.

chegewara, 2018-01-21:
Yes, esp32 and esp-idf sdk support classic bluetooth. Here you can find nice project:
https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

krisstakos, 2018-02-07:
Is it possible to detect/scan bluetooth classic devices somehow?

redplane, 2020-11-08:
3 years later, and I have the same question

chegewara, 2020-11-08:
This is BLE only library.

closed 2020-11-08 by chegewara

thinairart, 2022-02-27:
Just to follow up for future travelers who land here from Google:

CLASSIC_BT_ENABLED tells the BLE library to enable DUAL mode (BLE & SPP). This is important if you want the ESP32 to works as a dual mode device and support either connection type. You still need to use the Bluetooth classic library to actually implement Bluetooth classic SPP, and you handle that connection with that library, not this one. This flag just makes sure that Bluetooth is not enabled as BLE only, which would prohibit the SPP library from working correctly.